MarlinFirmware/Marlin #7675: how use g33 on delta?

EnricoSx, 2017-09-17:
hello, i try in my delta auto calibration with MANUAL probe
set delta heigh with m666 h301
execute g33
touch the paper
execute g33 until go to home
execute m500
`Calibration probing done.
	Calibrated 7 factors using 7 points, deviation before 3.3017 after 0.0367
	Endstops X-0.934 Y-1.770 Z0.000 height 307.825 diagonal rod 290.643 delta radius 159.371 Towers radius correction I6.56 J1.45 K0.00`

now my bed is Z7.8
if i try another G33 , the first point is out of bed.

can you teach me how to use it? thank you

LVD-AC, 2017-09-19:
Are you sure you're using Marlin FW? What version?

EnricoSx, 2017-09-19:
MalinKimbra MK4Duo, i know is similar to Marlin

LVD-AC, 2017-09-19:
The delta calibration is completely different in MarlinKimbra (a variant of Rich Cattell Marlin maybe?)

EnricoSx, 2017-09-19:
Ok, thanks

LVD-AC, 2017-09-19:
You can give the Marlin 1.1.5 from here a try; there's some errors in the MarlinKimbra and Rich Cattell calibration algorithms.

EnricoSx, 2017-09-19:
I contact magokimbra

LVD-AC, 2017-09-19:
close issue

closed 2017-11-14 by thinkyhead