MarlinFirmware/Marlin #7676: Want to use M600 with the Filament Sentinel, but can't use ADVANCED_PAUSE_FEATURE

eagleapex, 2017-09-17:
### Compile Error
I need to use M600 with what was formerly the FILAMENT_CHANGE_FEATURE. This is to trigger M600 for use with a filament sensor [here](https://dyzedesign.com/support/filament-sentinel-technical-support/#1487788661810-9ce9dcba-626e).
They do use a RC7 instead of the stable release.

The problem is that using ADVANCED_PAUSE_FEATURE requires an LCD which I don't have, and forcing in  FILAMENT_CHANGE_FEATURE to Configuration_adv.h causes a SanityCheck.h error. 

What are my options?

eagleapex, 2017-09-17:
I set the runout script to M112 so the printer will ESTOP if it runs out of filament. I'd still like to use a M600 script to replenish the filament.

Tannoo, 2017-09-19:
>They do use a RC7 instead of the stable release.

Who is they?

Try 1.1.x and see if it works there. RC7 is no longer supported.

ebraiman, 2017-11-20:
eagleapex,

I've been able to configure the filament runout with both a optical sensor and switch sensor in version 1.1.6 of Marlin.  Here are the steps I followed;

Mechanical endstop:
https://youtu.be/ijAuX8d3Mr4

Optical Endstop:
https://youtu.be/PiojWy5PDZE

There does appear to be some issue with the LCD and type used.  Reprap Smart Discount LCd appears to work better with optical endstop, whereas other issues occur when using mechanical endstop with Reprap Full display, when a SD drive is used.   The message thrown is M84 ... not M600 as it should be.  Granted I hope I'm configuring something incorrect, but another person would need to confirm.

Tannoo, 2017-11-20:
I use mechanical enstop for filament runout with RRFG display... no issues. I did change the FR pin to 11 instead of 4.

eagleapex, 2017-11-20:
The problem is I don't have a LCD panel. I got around it by making the
runout script just an Estop. Not the best, but safe.

Chris Thompson, eagleapex.com
"Obscurity is a far greater threat to authors and creative artists than
piracy" -Tim O'Reilly

ebraiman, 2017-11-22:
Tannoo,

Could you educate me on pin 4, as it appears not to work for anyone as far as I've seen on the forum using RAMPS 1.4. 

I've examined the pins_RAMPS.h and it appears pin 4 is used in another case ...

"#ifndef FAN_PIN
  #define FAN_PIN 4      // IO pin. Buffer needed
#endif

//
// Misc. Functions
//
#define SDSS               53
#define LED_PIN            13

#ifndef FILWIDTH_PIN
  #define FILWIDTH_PIN      4   // Analog Input on AUX2
#endif

// define digital pin 4 for the filament runout sensor. Use the RAMPS 1.4 digital input 4 on the servos connector
#define FIL_RUNOUT_PIN      5",

I'm not sure if the fan is related to BOARD_RAMPS_14_EFB or BOARD_RAMPS_14_EFF.  

Any ideas?

Regards,
Ed

AnHardt, 2017-11-22:
a.) FAN_PIN refers to D4 - FILWIDTH_PIN refers to A4
b.) As long an 'F' is in the type-string one of the 3 mosfeted outputs are used. I guess only for 'EEB' pin D4 is used.
c.) Using D4 for a hardware pwm is not a good idea (timer OC0B). Likely `FAN_SOFT_PWM` can help.

ebraiman, 2017-11-22:
AnHardt,

Sorry, "FAN_PIN refers to D4 - FILWIDTH_PIN refers to A4" was modified in my version, I swapped pin 5 and 4 and then used pin 5 for runout.  Want I was trying to say was, pin 4 does not work as documented for Fil_runout.  Maybe I could phrase it this way, has anyone on a RAMPS 1.4 configuration using pin 4 had runout work?

Regards,
Ed

Tannoo, 2017-11-22:
Yes. I do not use `FAN_PIN` or `FILWIDTH_PIN`.  So, I have pin 4 free for me. I use it for the `FIL_RUNOUT_PIN`.

Just sayin' that it works for me.

closed 2018-02-09 by thinkyhead