MarlinFirmware/Marlin #7677: Issue with large prints, firmware locks up and skips gcode

nucu, 2017-09-17:
Whenever I print objects who exceeds a certain size in the Y axis, some weird stuff happens:

> -the controller stops recieving gcode and just sends busy signals
> -when the last acknowledged gcode comannd is reached, **a lot of gcode commands are then skipped and just the E-Axis is driven a lot**, then at some point the firmware begins to accept and exec gcode and continues, but at a wrong point
> -also noticed that the displayed nozzle position is wrong and steady

### Infos:
Cartesian "BQ Hephestos 2" with "BQ_ZUM_MEGA_3D" board (Atmega2560)
Firmware: 1.1.4/1.1.5
The bug is maybe not limited to the Y axis, but my X axis is only 200mm long so i cant test it

### Tried:
-disabled watchdog -> no result
-diabled "PREVENT_LENGTHY_EXTRUDE" -> no result
-disabled Display to get more cpu power ->no result
-Octopi as well as SD-card -> error with both

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1309243/Config.zip)
Includes the two gcode files that fails and passes, but are nearly the same in size.
The Octopi Terminal log shows the point where "Send: N86" happens. Thats the last gcode that is send before the firmware refuses to accept more and also the last that is executed properly.

There were some similar bugs, but only related to delta printers:
#6703 #7303 #7510

Best regard, please be gentle with me, i dont do bug reports every day.

nucu, 2017-09-19:
I found the problem.

EXTRUDER_RUNOUT_PREVENT is not blocked during printing and will inject/add an extrude command into the planner. This will only happen when very long/slow moves are performed who exceedes EXTRUDER_RUNOUT_SECONDS.

thinkyhead, 2018-02-09:
See if it helps to change the conditions to these:
```gcode
  #if ENABLED(EXTRUDER_RUNOUT_PREVENT)
    if (thermalManager.degHotend(active_extruder) > EXTRUDER_RUNOUT_MINTEMP
      && !planner.blocks_queued()
      && ELAPSED(ms, gcode.previous_cmd_ms + (EXTRUDER_RUNOUT_SECONDS) * 1000UL)
    ) {
```
If so then we can use this as our solution.

boelle, 2019-02-19:
@nucu do you still have the problem with the latest bugfix 2.0 ?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-16:
i think there might be something to this issue

i tried to create a large cube 160x160x158.4 and print it as single wall

when print starts it probes bed like it should and heats up bed, but when bed it heated up it does not go to start pos. like the gcode says, but it tries to print outside the bed

i have marked with a red arrow in slic3r where it should start

[image]

boelle, 2019-03-16:
but it does start at the other side of the object
[image]

test code(rename from big cube test.gcode.txt to big cube test.gcode)
[big cube test.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/2974611/big.cube.test.gcode.txt)

boelle, 2019-03-17:
been printing a 160x160x158.4 cube in wase mode for 36 mins and so far i cant see any problem

closed 2019-03-17 by p3p

nucu, 2019-03-23:
The problem only occurs when a single gcode command is longer than EXTRUDER_RUNOUT_SECONDS. Vase mode is split in tiny high resolution gcode and will not trigger the problem.

#define EXTRUDER_RUNOUT_PREVENT 
#define EXTRUDER_RUNOUT_SECONDS = 10

G1 X0 Y0 E0 F900
G1 X200 Y200 E10

// 282mm total length with 15mm/s result in 18.8sec single gcode execution time
// 18.8sec > 10sec
// problem occurs, ring buffer gets corrupted, unexpected behaviour

I did not tested it since then because i dont use this feature anymore.