MarlinFirmware/Marlin #7678: Reprap discount full graphic controller and Velleman K8200 not working

Erlandsson, 2017-09-17:
- Description: ---

When flashing 1.1.4 bugfix with Arduino IDE 1.8.3 or 1.8.4 and u8glib 1.18.1 display is completely blank. 
(i have uncommented the full graphics controller line) and included the lib as needed.

So it tried:
1.1.0-r2  flashed with Arduino IDE 1.6.5-R5 the screen is garbled.

If original firmware Marlin 1.0.0-rc2 is flashed with IDE 1.0.5-R2 then the display works. (This is how i got firmware version which i never knew before, because it showed in the display).
(with the same u8glib as above.)

I need the 1.1.4 bugfix because i have added a filamentwidth sensor (and bedlevelling servo which works also on my old 1.0.0-rc2). 

Am i missing something or is the display on K8200 not compatible anymore in firmware? Or is it u8glib?
Or something more is needed to be configured?

rafaljot, 2017-09-18:
I tried #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
with Due

first issue was, I had to define:
#ifndef vsnprintf_P
#define vsnprintf_P vsnprintf
#endif

Unfortunately after success in compilation LCD doesn't work.

Erlandsson, 2017-09-18:
Well all of my above configs compile without problems, just the display is blank.

fiveangle, 2017-09-18:
Works fine on my system.  w/o knowing your config it's impossible to help.

Erlandsson, 2017-09-19:
**fiveangle** : you have a k8200 with marlin 1.1.4 or so and a reprap discount full graphic display working? 
No special firmwaremods to make it work?
The only addons i have is a servomotor for levellingarm which is connected to SS on the board (not used by display). And a filamentwidth sensor which is soldered directly to the cpu ( no pins broken out)..

fiveangle, 2017-09-19:
Please provide the info requested in the issue template (the part you deleted when you opened this issue - you can review it again by clicking "New Issue" and reading the info needed, then just provide it here in this ticket (w/o opening another issue) )

Thanks,

-=dave

rafaljot, 2017-09-20:
Is it first time are you trying this LCD with Velleman board? Do you made cable for it, custom pinout for this board etc ? 
tip: you can find two types of graphic display. Some, Chinese  have reversed plugs.

Erlandsson, 2017-09-20:
The display works as i said with marlin 1.1.0-r2 on my board. Read the first post. i am adding my config in the next post.

Erlandsson, 2017-09-20:
### Bug Report
- Description: ---

Reprap discount full graphic controller not working with marlin 1.1.4-bugfix

If old firmware 1.1.0-rc2 is flashed with ide 1.6.5-r5, the screen is garbled.
If same firmware is flashed with ide 1.0.5-r2 the display works completely.

And as firmware 1.1.4 is not compatible with ide 1.6.5 or older i canät use that to flash.

I have followed this guide and also tried some of the tips in responses there. This guy had the same problems with garbled screen and that it worked on older firmware.
http://www.opensourcelab.salazarserrano.com/how-to-connect-successfully-a-reprap-discount-full-graphic-smart-controller-to-a-3drag-3d-printer/

- Expected behaviour: ---

Working display.

- Actual behaviour: ---

Blank screen, as in nothing at all.

- Steps to reproduce:
  - Do this
  - Do that

adding config files.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1317696/config.zip)

fiveangle, 2017-09-20:
> If original firmware Marlin 1.0.0-rc2 is flashed with IDE 1.0.5-R2 then the display works. (This is how i got firmware version which i never knew before, because it showed in the display).
> (with the same u8glib as above.)

Do you have a link to the working 1.0.0-rc2 you used ?  Was this from the mfr, or from here ?

fiveangle, 2017-09-20:
Reviewing your config, it's obvious what the issue is.  You tried to drop the customized Configuration*.h  @CONSULitAS created into a much later Marlin version which doesn't have the bits required to support the display.  The original made the changes directly in Configuration.h, but we can't do that for Marlin baseline, so those changes will need to be made in the pins_3DRAG.h file (the master pins file for pins_K8200.h).

```
Searching 286 files for "K8200_VM8201"

C:\Users\djohnson\Downloads\Marlin-K8200-K8200_stable_2016-11-18\Marlin-K8200-K8200_stable_2016-11-18\Marlin\Configuration.h:
   62  
   63  // VM8201 Display unit
   64: #define K8200_VM8201
   65  // K8204 Z axis upgrade rod and coupler -> TODO
   66  // #define K8200_K8204
   ..
  156  // Define this to set a unique identifier for this printer, (Used by some programs to differentiate between machines)
  157  // You can use an online service to generate a random UUID. (eg http://www.uuidgenerator.net/version4)
  158: #if ENABLED(K8200_VM8201)
  159    #define MACHINE_UUID "2b7dea3b-844e-4ab1-aa96-bb6406607d6e" // K8200 standard config with VM8201 (Display)
  160  #else
  ...
 1034  
 1035  // K8200: for Display VM8201 with SD slot
 1036: #if ENABLED(K8200_VM8201)
 1037  
 1038  //
 ....
 1116  // produce one step. Should be increased for high-resolution encoders.
 1117  //
 1118: #define ENCODER_PULSES_PER_STEP 4 // K8200_VM8201: four steps per encoder step
 1119  
 1120  //
 ....
 1122  // move between next/prev menu items.
 1123  //
 1124: #define ENCODER_STEPS_PER_MENU_ITEM 1 // K8200_VM8201: One step per menu item
 1125  
 1126  /**
 ....
 1342  //#define SAV_3DLCD
 1343  
 1344: #endif // K8200_VM8201
 1345  
 1346  //=============================================================================

6 matches in 1 file
```

@CONSULitAS - are you interested in tweaking the current `pins_3DRAG.h `file to include the necessary bits required to support the VM8201 display for current Marlin ?

-=dave

fiveangle, 2017-09-20:
Here is current pins content:

```
//
// Misc. Functions
//
#undef SDSS
#define SDSS               25

#undef SD_DETECT_PIN
#define SD_DETECT_PIN      53

//
// LCD / Controller
//
#if ENABLED(ULTRA_LCD) && ENABLED(NEWPANEL)
  #undef BEEPER_PIN
  #define BEEPER_PIN       -1

  #undef LCD_PINS_RS
  #undef LCD_PINS_ENABLE
  #undef LCD_PINS_D4
  #undef LCD_PINS_D5
  #undef LCD_PINS_D6
  #undef LCD_PINS_D7
  #define LCD_PINS_RS      27
  #define LCD_PINS_ENABLE  29
  #define LCD_PINS_D4      37
  #define LCD_PINS_D5      35
  #define LCD_PINS_D6      33
  #define LCD_PINS_D7      31

  // Buttons
  #undef BTN_EN1
  #undef BTN_EN2
  #undef BTN_ENC
  #define BTN_EN1          16
  #define BTN_EN2          17
  #define BTN_ENC          23

#else

  #define BEEPER_PIN       33

#endif // ULTRA_LCD && NEWPANEL
```

Which matches @CONSULitAS branch:

```
//
// Misc. Functions
//
#undef SDSS
#define SDSS               25

#undef SD_DETECT_PIN
#define SD_DETECT_PIN      53

//
// LCD / Controller
//
#if ENABLED(ULTRA_LCD) && ENABLED(NEWPANEL)
  #undef BEEPER_PIN
  #define BEEPER_PIN       -1

  #undef LCD_PINS_RS
  #undef LCD_PINS_ENABLE
  #undef LCD_PINS_D4
  #undef LCD_PINS_D5
  #undef LCD_PINS_D6
  #undef LCD_PINS_D7
  #define LCD_PINS_RS      27
  #define LCD_PINS_ENABLE  29
  #define LCD_PINS_D4      37
  #define LCD_PINS_D5      35
  #define LCD_PINS_D6      33
  #define LCD_PINS_D7      31

  // Buttons
  #undef BTN_EN1
  #undef BTN_EN2
  #undef BTN_ENC
  #define BTN_EN1          16
  #define BTN_EN2          17
  #define BTN_ENC          23

#else

  #define BEEPER_PIN       33

#endif // ULTRA_LCD && NEWPANEL
```

So it seems most of the bits should be there, except the stuff in Config.  

The info regarding changing the optimization for the RR lib may have merit:

http://www.opensourcelab.salazarserrano.com/how-to-connect-successfully-a-reprap-discount-full-graphic-smart-controller-to-a-3drag-3d-printer/#comment-3

> Simon, I had a similar problem and solved it by disabling one line at "ultralcd_st7920_u8glib_rrd.h" file, the line is "#pragma GCC optimize (3)" located at the beggining of the file.
> I found something about this optimize messing with the LCD so I got lucky on the fix.

Have you tried commenting out Line 43 of ultralcd_st7920_u8glib_rrd.h and compiling with 1.8.4 to see if the LCD functions ?

-=dave

Erlandsson, 2017-09-20:
fiveangle: I have no idea to whom you are referring above.. Me or someone else?

The rc1.0.0-rc2 (i think it is because that is what the display show when workning) is an old k8200 version i now hav no idea to where it came from. It's 4 years ago i think it was installed. I probably got it from velleman's page or someting.

The VM8201 code you refer to is not used in my working code. I used the code for "reprap full graphic controller" further down in the same file.

**The disabling of "ultralcd:st xxxxxxx" have i also tried**. No difference. 

What confuses me in **above pins** code it that it does not use the same names as elsewhere in the code where it is called "**reprap discount fullxxxxx**" and all pins are spread all over the place.. 
I am no coder so i have problems understanding all.

I will try and remove all installations of everything and retry..

Erlandsson, 2017-09-20:
i will post the old working configs tomorrow.

fiveangle, 2017-09-21:
The ULTRA_LCD, NEWPANEL are overloaded to encompass the common features of a bunch of other displays of the same family (like the RRFGC) so it's very confusing, yes.

If you can provide the zip of the entire src tree (with your configs) that was working, that may be more helpful.

Erlandsson, 2017-09-21:
The working code does not compile with IDE 1.8.4, i get a lot of errors.
Here are my working complete old marlin:
[Marlin-K8200.zip](https://github.com/MarlinFirmware/Marlin/files/1321016/Marlin-K8200.zip)

Erlandsson, 2017-10-01:
Anybody ? No more info?

fiveangle, 2017-10-01:
You didn't respond what the result of the previous suggestions (changing optimization as per the link, and increasing delays).  Did you try ?

Erlandsson, 2017-10-01:
Oh.. you mean the: 
"Have you tried commenting out Line 43 of ultralcd_st7920_u8glib_rrd.h and compiling with 1.8.4 and the standard U8glib library accessible from within the Arduino IDE (not any downloaded/customized version) to see if the LCD functions ?

Another consideration is the possiblity that these timings have become non-optimal and may need tweaking ?"

No.. as you asked me to provide the zip for the old code i used... 
I really am not good at code at all..

For example:
#define CPU_ST7920_DELAY_1 DELAY_0_NOP

Do i change the second 0 to 1 or 2, or do i change delay_1 to delay_2 ?? I don't understand what to change..

Erlandsson, 2017-11-13:
I give up on this now.  I have spent too much time fiddling around, still won't display on new firmware, but is ok on old. So i migrate to ramps now.

closed 2017-11-13 by Erlandsson