MarlinFirmware/Marlin #7679: Puma 560 kinematics support?

rvsh2, 2017-09-17:
It would be great if you could add puma kinematics support in marlin firmware. This could lead to use the firmware in many more applications than 3d printing (for example pick&place robot arms).
Is there any chance to add this?

Thanks in advance!

thinkyhead, 2018-02-09:
It's not likely to happen anytime soon without more demand.

rvsh2, 2018-02-09:
It would be first open source puma kinematics /robotic arm 5/6 axis/ g-code supported firmware. It would beat repetier for sure :P
I could donate for this feature.

thinkyhead, 2018-02-17:
If you can donate a Puma robot, that would get the ball rolling.

thisiskeithb, 2021-12-29:
Now that Marlin supports up to 6 axis (with support for up to 9 pending), does that cover this feature request?