MarlinFirmware/Marlin #7680: Allowing homing for XYZ or only Z only if nozzle it heated

RajaaKahel, 2017-09-18:
Hi, 
Is there a way to allowing homing for XYZ (or only Z) only if nozzle it pre-heated to a certain temperature?
I'm planing to put 3 metal pads on the glass to use with the nozzle as electrical-contact sensor to work as Zendstop and as 3-point auto-leveling. Because if there are some cold plastic on the nozzle tip or on the little pads, there will be no electrical connection.. So if the nozzle is heated it will be more safe (even after cleaning the plastic).
Thank you!

fiveangle, 2017-09-18:
Honestly, w/o a scrubbing (and heating) mechanism similar to how Lulzbot implemented, it will not work.  I hate to desuade you, but is there a reason you don't want to go with something a bit more standard like bltouch ?

RajaaKahel, 2017-09-18:
I think that mechanism could be easily made, but I also think that even if there is some plastic, if the nozzle is hot, it'll make contact. I'm going to make an experiment tommorow to confirm this.
I don't like BLtouch, it's not the best accuracy, it needs offsets, it runs into issues sometimes.
Also I like DIY stuff and experiments :)

RalphSch, 2017-09-21:
I'm already using the direct nozzle touch method for some weeks on an precision aluminium plate and because of the of the offsets of 0 it's perfect for UBL because it can reach every point of the bed at a very high precision (without an additional deployment movement). You do not need to wipe the nozzle which just squeezes the molten filament away. Also it's simple (no need for additional IO like bltouch) and cheap (my 6mm plate costs about 10€) ..... but you must have a clean or heated nozzle so this feature requests would prevent one of the greatest problems with this method.

RajaaKahel, 2017-09-21:
RalphSch, exactly what I think. So do you have Endstop for Z as well? Or you never home with cold nozzle?
Can you please send me the link for the place you bought the aluminum bed from?
Thanks

RalphSch, 2017-09-22:
I think (don't have looked deeper yet) that mostly every part of needed code is already given because the heatup procedure is also needed and included in the filament change feature. May be get some time over the weekend to have a deeper look in the code.

Little bit offtopic:
I don't have an additional Z-Endstop but I should definitely add one because I once again bend my Y rods too much and also did some engravings on the aluminium plate with my nozzle (which works quite well :D ). The plate is from an german online metal online shop (https://metall-fachhandel.de/produkt/plangefraeste-aluminium-gussplatte/), cutted but not drilled. You should find a similar store in mostly every country or just ask local metal shops. The material used by the german shop seems to fit perfectly for a heatbed (thanks to DeepL.com for translating):
Aluplan is a cast, homogenized (stress-free annealed) plate of the aluminium alloy EN AW-5083, which is milled with high precision on both sides, the solution for applications requiring the aluminium alloy EN AW-5083 and at the same time demanding high flatness, dimensional stability and surface quality.
Max. Dimensions for shipping: 2500mm x 760mm - Larger dimensions can be requested directly by mail.
Cutting tolerances: -0mm / +0,5mm
traits
Very good stability
Extremely low internal tension
Best possible flatness after sawing guaranteed
Excellent corrosion resistance
Excellent anodic stability
Ideal welding behaviour
Good mechanical properties
Tight thickness tolerance
Minimum porosity
Very smooth surface
Both sides with protective foil

RajaaKahel, 2017-09-22:
Awesome, thank you! :D

boelle, 2019-02-19:
@RajaaKahel so you got the feature or ??

RajaaKahel, 2019-02-19:
@boelle
I actually moved to DuetWifi with RRFW, now I can do whatever I want :)
With all respect and love to Marlin.

boelle, 2019-02-19:
oki, but its still a usefull thing now that stepper drivers like TMC can detect if nozzle touches the bed

@RalphSch how do you do it?

RajaaKahel, 2019-02-19:
@boelle
I have these drivers on my Duet, but this doesn't actually give high precision (not recommended for Z).
I bought the aluminium plate from the website that RalphSch suggested, works great. I didn't even need auto-leveling sence then.

ellensp, 2020-08-24:
Requester moved platforms.  no longer relevant.

closed 2020-08-24 by ellensp