MarlinFirmware/Marlin #7681: Arm Support

Penagwin, 2017-09-18:
Could I get an overview on roughly how to setup marlin on the re-arm or the due? Which would be better, the re-arm? Is this a viable setup right now (Stable)?

There are a ton of branches and I don't know which is best. I'm sorry for the dumb questions.

fiveangle, 2017-09-18:
It is still under very active development with Re-ARM most current.  Please close this issue and review thse threads #7121, #7076, #7622, #7511, #6524, https://github.com/MarlinFirmware/Marlin/pull/7028#issuecomment-310880218, #7264

Welcome aboard ! :)

-=dave

Bob-the-Kuhn, 2017-09-18:
The Re-ARM code isn't ready for prime time yet.  Some people have reported success, others have had problems.  You'll need to use PlatformIO to compile the code.

For the Due you should be able to compile Marlin 1.x using Arduino.

There's also the [MK4duo](https://github.com/MKFirmware/MK4duo) fork.  It's basically Marlin 1.x but with an online configuration tool.  It works nicely for most printers.  There are definitely option combinations that it doesn't support.

If you need 6 stepper drivers then a Due and the RAMPS-FD is a nice combo.

closed 2017-11-14 by thinkyhead