MarlinFirmware/Marlin #7682: Bad spiraling circles

Leenderm, 2017-09-18:
### Bug Report
- Description: --- Large circles progressively warp into ovals spiraling up the print see pic
- Expected behaviour: --- Circular circles
- Actual behaviour: ---  Large circles progressively warp into ovals spiraling up the print
- Steps to reproduce:
 CoreXY printer, print tall round vase, seems to occur when Z is greater than 256mm
[image]

Tannoo, 2017-09-19:
I will say, that is impressive. Too bad I can't replicate that.

fiveangle, 2017-09-20:
Agree - that is awesome !

Leenderm, 2017-10-01:
The CoreXY forum helped identify the problem, it was the print flexing from the extruded, once printed 1.1mm thick it printed fine.
[image]

closed 2017-10-01 by Leenderm

Tannoo, 2017-10-02:
Oooh... that's even nicer!