MarlinFirmware/Marlin #7684: UBL Fine Mesh Bed leveling adjustment steps

devils27, 2017-09-19:
Hi, I have set up my UBL mesh but when trying to do the G29 P4 on a point I am limited in the amount/steps to increase or decrease a point to adjust it.  If it is initially set by the probing to be 0.054, if I turn the knob to decrease it goes to 0.050 and if I increase it goes to 0.060, but if I try to go decrease again I cannot get back to 0.54, only 0.050.  Is there a way to configure Marlin to allow fine tuning to be less steps instead of being 0.01? I would like 0.001 steps since that is what probing does anyway.  I did it with manual mesh bed leveling to allow "finer" steps when manually leveling, how can I do this with the fine UBL leveling?  Thanks.

Also, Is there a way to manually input point numbers for a UBL Mesh?  I copied all of my 5x5 mesh point numbers down.  If I were to get a new board or erase EEPROM is it possible to manually input all of my UBL mesh point numbers as to not have to go through the whole probing and fine tuning again? (issue is when fine tuning you have to put the nozzle at a specific point and sometimes it may not be exact so inputting numbers that way will be inaccurate)

Thanks!!!!

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Roxy-3D, 2017-09-19:
The step size is mostly limited by how many encoder clicks you have per menu item.   

It might be good to have a way to dynamically adjust this.  But for right now...   Make it so it takes more encoder clicks per menu item.   And this will carry over into the G29 P4 editing.

devils27, 2017-09-19:
How do I change encoder clicks per menu item.  every turn of the knob goes increases or decreases 0.010 for ubl mesh fine tuning but only for that from what i can see.  Is it at the bottom of the configuration.h somewhere that I am not seeing?

devils27, 2017-09-19:
Is that "//#define ENCODER_STEPS_PER_MENU_ITEM 5"?  how would that correlate to ubl fine mesh steps when fine tuning?

devils27, 2017-09-19:
Great thanks, i saved mesh to gcode that I can run if i need to reload it.  As for the steps, i found that I can modify one of the .cpp files but that might make the firmware unstable, plus leveling to that degree probably wont make a whole lot of difference and/or able to be kept that strict.  ty

closed 2018-02-09 by thinkyhead