MarlinFirmware/Marlin #7685: G-code/Gcommand Parser Updated

kcheeeung, 2017-09-19:
Hello friends once again,
I have been using a modified variant of the Marlin for some time now, and I haven't been keeping up with the development of the Marlin firmware. If I recall correctly, the Marlin firmware's gcode parser was updated for better management, and it seems that I have lost some of the abilities written in my Marlin when I tried to update to the new firmware with my custom commands.

The latest version of my Marlin firmware was RC8, which was some time ago. This code in Marlin firmware RC8 works **without any problems.**
https://github.com/kcheeeung/Marlin/blob/master/Marlin/Marlin_main.cpp#L7648
```
inline void gcode_M430() {
  int pulse_usec = 200; //Default to 200 usecs
  if (code_seen('S')){
    pulse_usec = code_value_int();
  }

  if (!code_seen('V')) return; //You must pick a valve number
  int ValveNumber = code_value_int();
  if (ValveNumber == 1){
    pinMode(VALVE_1_PIN, OUTPUT);
    WRITE(VALVE_1_PIN, HIGH);
    delayMicroseconds(pulse_usec);
    WRITE(VALVE_1_PIN, LOW);
  }
}
```

In updated Marlin 1.1.5, I pasted the same code and received an error. This I believe is due to the updated gcode parser, and the way the Arduino interprets the code. **How would and where would I then have to update or change the Marlin firmware to make my code work? Ex. how would I make my code on the top work?** Thank you so much.
```
Marlin_main.cpp:9200: error: 'code_seen' was not declared in this scope
   if (code_seen('S')){
                    ^
Marlin_main.cpp:9201: error: 'code_value_int' was not declared in this scope
     pulse_usec = code_value_int();
                                 ^
```

teemuatlut, 2017-09-19:
code_seen() -> parser.seen()
code_value_int() -> parser.value_int()

Refer to ```gcode.h```.

fiveangle, 2017-09-19:
You can also disable the new gcode parser, which may allow you to abandoned trying to fix/update the new parser, instead using the old parser with your customizations.  It's in the advanced config.

-=dave

kcheeeung, 2017-09-20:
Thanks for the help @teemuatlut!

closed 2017-09-20 by kcheeeung