MarlinFirmware/Marlin #7688: [Marlin 2.0 Bugfix Branch new layout] - Compile Error with BLTOUCH and PROBING_HEATERS_OFF

DaveAhrendt, 2017-09-19:
Compiling for MOTHERBOARD BOARD_RAMPS_14_EFB
With BLTOUCH and PROBING_HEATERS_OFF defined compiling fails with:

Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Stuff deleted...

C:\Users\Dave\AppData\Local\Temp\arduino_build_63964\sketch\src\module\motion.cpp: In function 'void do_homing_move(AxisEnum, float, float)':

C:\Users\Dave\AppData\Local\Temp\arduino_build_63964\sketch\src\module\motion.cpp:888:43: error: 'probing_pause' was not declared in this scope

     if (axis == Z_AXIS) probing_pause(true);

                                           ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_63964\sketch\src\module\motion.cpp:908:44: error: 'probing_pause' was not declared in this scope

     if (axis == Z_AXIS) probing_pause(false);

                                            ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Phr3d13, 2017-09-19:
I may be wrong, but I think 2.0 is only 32bit at the moment.

fiveangle, 2017-09-20:
No, the flood gates were just opened for AVR issues in the 2.0 "new layout" branch.

Prepare yourselves.  Bug reports are coming... :)

But you will need to provide your configs for anyone to help resolving your compile error (it shoudl have shown you a template with this requirement when you opened the issue.

DaveAhrendt, 2017-09-20:
Just use the default config files from the release and uncomment the BLTOUCH and PROBING_HEATERS_OFF #defines and try to compile.  My specific config files are irrelevant in this case.

fiveangle, 2017-09-21:
Drop this probe.h into your tree:

https://github.com/fiveangle/Marlin/blob/bf20ng-undefined_probe_fn_fixes/Marlin/src/module/probe.h

fiveangle, 2017-09-21:
#7699 has been merged; please update your local and test/close if resolved.

Thanks,

-=dave

PR #7699 (2017-09-21): [bf20ng] Resolve undefined references to probing functions compile errors
Resolves #7688 

[Scott, is there some more elegant way to resolve this ?  It appears to be fallout from breaking these out from Marlin_main.cpp]
closed 2017-09-25 by DaveAhrendt