MarlinFirmware/Marlin #7689: I need a working config

eMotion2, 2017-09-19:
I tried so much now to get my bed leveling working. Does anybody has a config file which works for an induction probe to do bilinear auto bed leveling?

guiseco, 2017-09-20:
What kind of problem do you have ?

Tannoo, 2017-09-20:
Please follow the template that you deleted. It will help us help you.

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

eMotion2, 2017-09-20:
I already posted a detailed issue. ( https://github.com/MarlinFirmware/Marlin/issues/7652 )
However I thought now that it may be easier just to get a config which works for probing and ABL and then adapt it to my printer.

Tannoo, 2017-09-20:
Well, just configure it for your printer. That will be faster.

eMotion2, 2017-09-21:
Not really becuase whatever I change The problem I have stays, so I thought I try a working config. So if somebody has a **Working Config.zip an Induction Probe ** give it to me pleaaaase.

Tannoo, 2017-09-21:
Download and try the latest `bugfix-1.1.x`.

Tannoo, 2017-09-21:
If you tried to run my configs, you would likely get compile errors.

eMotion2, 2017-09-23:
Thanks for every help, I got it mostly fixed now. Therefore I will clse this 😃

closed 2017-09-23 by eMotion2