MarlinFirmware/Marlin #7693: display settings for velocity don't change

ruggb, 2017-09-19:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

ruggb, 2017-09-19:
bugfix-1.1.x-ca97348
the velocity max shown on the display is 200 for x&y. I change it in config to 500 and upload it and it remains 200. 
Text says if eprom is enabled the saved settings override the config. But I don't see any eprom enable any more.
I change it on the display and save settings and I get a beep (I think it is an error beep) but nothing happens. I go back and look and it is still 200. 
What am I missing?

ruggb, 2017-09-19:
never mind. I rebooted everything and it is back to "normal"
thx

fiveangle, 2017-09-20:
close then ?

ruggb, 2017-09-20:
sorry

closed 2017-09-20 by ruggb