MarlinFirmware/Marlin #7694: instruction vs code

ruggb, 2017-09-20:
1.1.4-bf-386b74f

line 516
 * Note that if EEPROM is enabled, saved values will override these.

but I searched for an eprom enable and can't find it anymore. It used to be there.
Is it missing in action or does the text need fixing?

Tannoo, 2017-09-20:
In `Configuration.h`:

```
//=============================================================================
//============================= Additional Features ===========================
//=============================================================================

// @section extras

//
// EEPROM
//
// The microcontroller can store settings in the EEPROM, e.g. max velocity...
// M500 - stores parameters in EEPROM
// M501 - reads parameters from EEPROM (if you need reset them after you changed them temporarily).
// M502 - reverts to the default "factory settings".  You still need to store them in EEPROM afterwards if you want to.
//
//#define EEPROM_SETTINGS // Enable for M500 and M501 commands
//#define DISABLE_M503    // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT   // Give feedback on EEPROM commands. Disable to save PROGMEM.
```

ruggb, 2017-09-20:
OK, so you removed the text about ENABLE EPROM support 
//define this to enable EEPROM support
//#define EEPROM_SETTINGS

and added this specific
//#define EEPROM_SETTINGS // Enable for M500 and M501 commands

but left this
* Note that if EEPROM is enabled, saved values will override these.

If //#define EEPROM_SETTINGS is to enable eprom for everything then why tag it with this?
// Enable for M500 and M501 commands

May I suggest this change
* Note that if "#define EPROM settings" is uncommented, saved values will override these.

Or put this back in
//define this to enable EEPROM support

Phr3d13, 2017-09-22:
Once the dust from the new 2.0 layout changes settles, I'll edit that line for clarity and make a PR. Maybe something like this would suffice?
`//#define EEPROM_SETTINGS // Enable for EEPROM support, such as M500 and M501 commands`
Would this make things more understandable?

ruggb, 2017-09-22:
Not sure why M500 & M501 need to be called out specificallly. It worked the way it was. = "eprom support" covers it all. Whatever.

Phr3d13, 2017-09-23:
So people know what to look for when M500 doesn't work for them?

ruggb, 2017-09-23:
OH, those are the save functions, got it.

Tannoo, 2017-09-23:
M502 shouldn't work either if `EEPROM_SETTINGS` is not enabled.

Tannoo, 2017-09-23:
Well... maybe so.

closed 2017-11-27 by thinkyhead