MarlinFirmware/Marlin #7696: [FR] Marlin Resource Optimization Guide

Deneteus, 2017-09-20:
FWD: This is my first time compiling Marlin. 10+ years of QA experience with software/hardware.

Issue:  When I was attempting to compile Marlin for my Creality CR10 with Unified Bed Leveling (UBL) enabled I quickly found that it eats up a lot of space. 

* I already had a copy of 1.1.4 that was compiled that I had sources to from 2 different sources and realized that neither source included any notes about the changes they had made so there was no way to know the reasons behind the changes. For someone working with open source products I thought 'That's not cool man.' How is a noob supposed to learn how to modify this stuff? 
 
I then scoured the internet for more information about this issue and eventually found several posts pointing to the fact that the code required to enable the feature on my Creality CR10 would require me to sacrifice features that were already enabled in the stock firmware without Auto Bed Leveling (ABL)  such as SD Card support, LCD Display Support, Sound, etc.

I found that there is an issue with the Rerapdiscount Full Graphic Smart LCD using 55kB as well as the fact that previously some changes could be made to QR_SOLVE.H to reduce the amount of memory as well.  

As I read through previous bug reports I found that that there are a few people that know this info but I saw none of this info within the documentation itself where anyone could figure it out themselves. When there are 20 Instructables about an issue but it's not documented in official documentation then I believe there are a few problems that need to be solved.

1. Why isn't there any info about how much memory is required to run different types of LCD hardware?
2. Why is there no list of 'Most Wanted' optimizations being posted so that new people can help find better ways to fix some of these issues?
3. Where are the docs that say 'if you do this' then you are going to bork your install? 

Feature Request: Please provide more information on managing available memory when enabling/disabling Features within Marlin within the FAQs. It would be helpful to include the average size of any compiled drivers for LCDs or anything else that's going to use space when the firmware is compiled. This way people can plan their builds around available resources.  

Supporting Documentation: 

Graphic LCD support makes Marlin too big for 128k boards.
https://github.com/MarlinFirmware/Marlin/issues/1645

V1.1.4 with Anet and Bed Leveling Sensor -> "Sketch too big;"
https://github.com/MarlinFirmware/Marlin/issues/7240

Marlin with Anet A8
https://github.com/MarlinFirmware/Marlin/issues/7190

Not compiling with UBL defined #6602
https://github.com/MarlinFirmware/Marlin/issues/6602

fiveangle, 2017-09-20:
Looks like you found lots of info on this (as per your links above).  I answered you in your other thread too.

The reason there isn't specific answers on how much each take, is because they are not constant but instead depend on all the other enabled features.  For example, enabling LCD but not UBL or SD does not increase PROGMEM usage as much as when UBL and SD are enabled (because without them, the LCD logic to run those features isn't needed).  With dozens of compile-time features in Marlin, and every feature adding another increment to the exponent of how many possible outcomes there are, the answer becomes very complex, very quickly.

Scott and I discussed the possibility of scripting a travisci compile for every possible perturbation (perhaps on a daily basis, since it would be quite resource-consuming) and auto-populating the results in yet another as-yet-to-be-built "_Marlin Configurator_" that would cite PROGMEM/SRAM usage ranges for the possible outcome, but the reality is that doing this would amount to almost as much effort as most of the actual dev work being done on Marlin itslef as of late.

I like your idea though and absolutely agree that it would be a great thing to have for the small number of users that it is critical for.  If you have insight as to how to go about doing so, please jump in, but providing the info is much more difficult than it seems on the surface, with a relatively small number of users it benefits.

The configure->compile->review resulting size of sketch method that everyone has been using for years, while somewhat tedious, seems to work well enough that the drive to ochestrate a more elegant solution hasn't spured anyone to want to invest the time to do so.

All this said, I do think a simple document expaining the _configure->compile->review resulting size of sketch method_ would be useful for new users struggling with where to start.  Since you're headed down this path, perhaps you would be interested in writing such a doc ?  It would make a nice addition to the Marlin wiki and, as you're finding, would help a lot of new users out.

-=dave

Deneteus, 2017-09-20:
@fiveangle Sorry. I didn't see your response in my email this morning. I actually wrote that this night. This surely needed to be its own Feature Request. 

I figured that was the case with the PROGMEM usage by the random hints about what could be disabled. 
I played around with it for a couple days myself so when you say 'a relatively small number of users'. I am looking at everyone with 128k or less which I would figure is alot of users. 

Earlier this month I had an issue with my Wanhao i3 Duplicator Plus where I was trying to print with ABS and the printer was shutting down the hot end and bed at the same time with no error messages being displayed on the panel and no errors being logged in Cura (because there are no error logs in Cura apparently when a print fails.)  I had already upgraded to their V3.0 firmware and it corrupted my LCD images so I contacted support and they sent me another firmware that had not been released that also was not working. 

I literally spent about a month trying to figure out what was wrong. I dug through 2 years worth of Google Group support posts. I contacted Wanhao support and wasn't getting anywhere when I decided to try i3 PlusPlus based on Marlin 1.1.1 and it did the same thing and but when it did I received an error message about the Temp Runaway and realized that the suppressed the error message by not including on in their GUI.

[image]

So that meant the entire time I was having an issue it was because Wanhao set the temps in Marlin where they couldn't be near 255C. So the first thing I did was contact them to get the source so that I could modify it myself and instead they sent me a firmware compiled for 280C.

If I had the source from the very beginning then I would have figured this out in like 10 seconds.

I think the 'configure->compile->review resulting size of sketch method' in this case is like building a car and then realizing that because the engine is too big that the headlights won't fit.

I actually just setup Octoprint for both of my printers a few weeks ago but sometimes you need the card for other people to bring you prints and you don't want to have to rely on a laptop/network to get your prints out. It would be cool if you could use the USB port to do that.

I would be interested in writing such a document but I don't think I have enough info to reliably cover each of the features. I'm essentially just getting started. I would think the example configs that you already built with the memory usage would be something you could add to the example configs folder on a regular basis.

Roxy-3D, 2017-09-20:
@Deneteus   It really shouldn't be that much work to make UBL fully configurable so the user can pick and choose which portions of it they want.   G26 already can be turned off.  And UBL can now be used without an LCD Panel in the system.        We can go further and make it so the interactive mesh editor and automatic probing are able to be turned off.   Or maybe one can be turned on to get the initial mesh and then turned off to make room for the interactive mesh editor later.

None of doing that is 'Rocket Science'.   I did have it running on a 128KB PrintRboard 3 or 4 months ago.   It should be fairly straight forward to get UBL to work on a 128KB AVR board with a 20x4 LCD Panel.    **_If you want to see how far you can shrink the UBL foot print I'll help coach you to do that._**

Deneteus, 2017-09-20:
@Roxy-3D 

* I want to keep all the useful features for my specific printers and disable anything that is not going to make the prints better, or that will help them take less time to print. I am ready when you are. 

* I want to keep LCD but I want to reduce its footprint if there are any unnecessary features. I have been looking at the posts involving u8glib and I understand that it is a lost cause.

* I have developed patience for this type of stuff but most new users do not. That is the only time where Rocket Science comes in.  I could have disabled stuff and went about my merry way but it wouldn't do anything to help anyone else who is just starting out with lowest common denominator hardware.

* I am eventually going to upgrade to the CR-10S board and I am going to build another printer using the STEVAL-3DP001V1, assuming they don't discontinue it. The Duet Wifi seems overpriced for how long it has been out.

Roxy-3D, 2017-09-21:
@Deneteus 

It should only take serveral #define's to be added to control features to make it possible to put UBL into a 128KB part.    My initial suggestion would be to add:
- #define UBL_NO_MANUAL_PROBE
- #define UBL_NO_LCD_SUPPORT
- #define UBL_NO_J_MESH_TILTING
- #define UBL_NO_EXTRANEOUS_MAP_STYLES   This one may need explanation.  I'm thinking G29 S -1 and G29 T 1 and G29 T 2 are not supported.
- #define UBL_NO_ WHAT_CMD
- #define UBL_NO_DEBUG_CODE  This one may need explanation.  There are a lot of places in the UBL code where debug functions are present to print out the current state of the machine.  For example, in ubl_motion.cpp there is void debug_current_and_destination().   Turning off debug code should save some space.

I bet if we get those #define's in place with the corresponding #ifdef's to control code generation, we will be where we need to be.   And if we aren't...  We should be close enough that we can find the remaining memory to free up so it fits.   My suggestion is we do the work on the new (when it gets dropped in place) bugfix-2.0.0 just because that is coming in 5 kb smaller on a fully optioned firmware build.   And...  that will be the current development branch too.

Deneteus, 2017-09-23:
That sounds like a plan.  I can always add an LCD to my Octoprint box so I can tell whats going on without walking to my PC.

fiveangle, 2017-09-23:
I put one of the Amazon Fire's that I got on black friday for 40 bucks over my printer for just this purpose.  Works great.

Deneteus, 2017-09-24:
I was just going to use one of those touch screen LCD's. Maybe a EPAPER 3 color.

Deneteus, 2017-09-27:
Could you take a look at my configs and tell me if I missed anything?

[Marlin 9.26.17 Bugfix Configs .zip](https://github.com/MarlinFirmware/Marlin/files/1335642/Marlin.9.26.17.Bugfix.Configs.zip)

Deneteus, 2017-09-27:
Enabled Safe Z Homing and found the Z was backwards from the Example file when homing.

thinkyhead, 2017-09-27:
The best starting-point for CR-10 configs is to use this version of Marlin and the CR-10 example configs that come with it:

https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip

It has the following extra features enabled:

- `LEVEL_BED_CORNERS`
- `PRINTJOB_TIMER_AUTOSTART`
- `QUICK_HOME`
- `STATUS_MESSAGE_SCROLLING`
- `SDCARD_SORT_ALPHA` (100 x 28 reserves 2.73k SRAM) … (256 x 28 reserves 7k SRAM)
- `BABYSTEPPING` (Z only, with 10x multiplier and double-click option)
- Disabled `ARC_SUPPORT` — Rarely needed, non-standard in all slicers.
- `ADVANCED_PAUSE_FEATURE` + `PARK_HEAD_ON_PAUSE` + `HOME_BEFORE_FILAMENT_CHANGE`
- `NO_WORKSPACE_OFFSETS` — `G92` simply uses old behavior.

UBL is too large a feature to fit onto the Melzi board. It only has 128K of flash (and 16K of SRAM). In fact, I can get no automated form of bed leveling to fit within the allotted space using Marlin 1.1 — though I didn't try turning off any features that I wouldn't want to sacrifice. It might be possible to squeeze Mesh Bed Leveling (which is manual) onto the board by turning off just about everything.

The "EZABL" version of Marlin is not as up to date as the link above, so it is able to fit a form of bed leveling. Older versions of Marlin are generally less efficient and not as refined, so it's best to use the most recent version that you can.

Deneteus, 2017-09-27:
I am already using last nights bugfix version and it fits because I disabled the LCD and a few other features.

Roxy-3D, 2017-09-27:
Also...   There was a separate request to make UBL work without a Z-Probe.   That would release all the program memory used to do the various probing functions.   I have helped people fake out the printer and bring up UBL without any Z-Probe.   It works, and is pretty straight forward especially if the bed is some what level.    You just do a G29 P0 to zero the entire mesh and jump straight into G26 and start editing the mesh.

The problem is...  To do that, you really need the G26 code.   Hopefully, we will be able to turn off enough stuff that G26 can still be turned on for people without Z-Probes on a 128KB processor.

fiveangle, 2017-09-27:
Just a cursory look at your configs and there's countless places to regain PROGMEM.

Your current config:
```
Sketch uses 107642 bytes (84%) of program storage space. Maximum is 126976 bytes.
Global variables use 3453 bytes of dynamic memory.
```

With some simple changes:

```
Sketch uses 95880 bytes (76%) of program storage space. Maximum is 126976 bytes.
Global variables use 3377 bytes of dynamic memory.
```

Of course still nothing close to enable even a text-based LCD.  The only option for bed leveling + LCD on 128k boards is bilinear.  Trust me, I've explored every option :)

If @Roxy-3D can untie UBL from automatic probing to get some progmem back, then you might be able to bring UBL up w/o and LCD with G26 validation enabled, then once that's done, disable G26 / re-enable LCD, and go to town.

Until then, I highly suggest enabling Linear Advance.  It will improve your print quality better than any other thing, especially with CR-10 if you plan to use any of it's massive size, otherwise you are stuck with 60-100hr for big prints on it.  Also, acceleration of 500mm/s is quite slow, especially without Linear Advance, so you really should bump that to 700-1000 for print speeds or you'll get bad bulges on corners, esp. with the long bowden on the CR-10.

fiveangle, 2017-09-27:
...complete aside - it astounds me that panels take soooooooo much space to implement.  I wonder how @apballard 's Nextion implemenation compares, since the display handles most of the heavy lifting ?

https://youtu.be/4NnOwrkCkYk
https://github.com/apballard/Marlin

When I'm not late for work (like now ;) I may compile it to see...

Deneteus, 2017-09-27:
Holy cow @fiveangle fiveangle. I was just thinking about doing the same thing with an external LCD last night. I could have it to send data to my Samung using an app. You could use a small Smart-tv with to manage multiple printers info that way just by adding touch.

Roxy-3D, 2017-09-27:
> Working without a Z-probe. What sorcery is this?

Well...   Instead of using a Z-Probe in the system, it would home using normal endstops.    And instead of usinga Z-Probe to do the initial probing of the bed, you just set the entire mesh to 0.00mm using G29 P0.     Obviously, those values are going to be wrong.  But you just jump straight into a G26 / G29 P4 R cycle and edit the mesh to the correct values.

closed 2018-02-06 by thinkyhead