MarlinFirmware/Marlin #7704: [Marlin 2.0 Bugfix Branch new layout] Anet compile issues

Phr3d13, 2017-09-21:
### Compile Error
When I compile with the anet a8 example config I get an error:
```

C:\Users\Phr3d\AppData\Local\Temp\arduino_build_182463\sketch\src\module\temperature.cpp: In static member function 'static void Temperature::isr()':

C:\Users\Phr3d\AppData\Local\Temp\arduino_build_182463\sketch\src\module\temperature.cpp:1975:33: error: 'START_ADC' was not declared in this scope

         START_ADC(ADC_KEYPAD_PIN);

                                 ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Error compiling for board Anet V1.0.

[Anet example configs.zip](https://github.com/MarlinFirmware/Marlin/files/1321230/Anet.example.configs.zip)

fiveangle, 2017-09-21:
Please provide configs that result in above error, and provide correct board selection that should be set in Arduino IDE

Phr3d13, 2017-09-21:
I added the board definition, and fixed the tagging of my previous post.

Phr3d13, 2017-09-21:
[(https://github.com/MarlinFirmware/Marlin/commit/6adab55f593637acd83856f32bd6c07ecf553ef7)] Fixes this.

Phr3d13, 2017-09-28:
Thinkyhead's recent commit fixes this.

closed 2017-09-28 by Phr3d13