MarlinFirmware/Marlin #7705: [Marlin 2.0 Bugfix Branch new layout] GT2560 example compile issues

Phr3d13, 2017-09-21:
### Compile Error
When I compile with Geeetech GT2560 example config I get an error (actually 2, but the first was just complaints about old version config, I edited the #'s after comparing the new default example and the new gt2560 example and not seeing any big differences) :
```
C:\Users\Phr3d\AppData\Local\Temp\arduino_build_251570\sketch\src\gcode\control\M80_M81.cpp: In static member function 'static void GcodeSuite::M81()':

C:\Users\Phr3d\AppData\Local\Temp\arduino_build_251570\sketch\src\gcode\control\M80_M81.cpp:106:13: error: 'suicide' was not declared in this scope

     suicide();

             ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

fiveangle, 2017-09-21:
Please provide the updated config that results in the above error

Phr3d13, 2017-09-21:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1321989/Configuration.h.txt)

Phr3d13, 2017-09-21:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1321992/Configuration.h.txt)

Phr3d13, 2017-09-21:
I don't think this is related to the long filename issue

Phr3d13, 2017-09-21:
Adding
 `  #ifndef suicide()
  #define suicide()
  #endif`
before 
`suicide();`
in M80_M81.cpp
allows this to compile, don't know if it will break anything though.

Phr3d13, 2017-09-21:
[This commit](https://github.com/Phr3d13/Marlin/commit/0555ded24f83aefe3633cf40138433904cadbb69l) looks like it fixes this.

Phr3d13, 2017-09-27:
Should I submit new PR's for the renamed fork (now bugfix-2.0.x)?

Roxy-3D, 2017-09-27:
Yes.  We are trying to merge everything back together again.   bugfix-2.0.x  is the main development branch (even if everything isn't 100% working yet).

Phr3d13, 2017-09-27:
[This](https://github.com/MarlinFirmware/Marlin/pull/7767/commits/33588081783cab7252f6fcff5d05b064ae6993fb) commit should fix this.

Phr3d13, 2017-09-28:
Thinkyhead's recent commit fixes this.

closed 2017-09-28 by Phr3d13