MarlinFirmware/Marlin #7707: [Marlin 2.0 Bugfix Branch ] GT2560 compile issues

vMeph, 2017-09-21:
Using library U8glib at version 1.19.1 in folder: C:\Users\vMeph\Documents\Arduino\libraries\U8glib 
fork/exec C:\arduino-1.8.4\hardware\tools\avr/bin/avr-gcc.exe: O nome de ficheiro ou a extensÃ£o Ã© demasiado longo.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

fiveangle, 2017-09-21:
Potentially dupe of #7705

Please provide configuration*.h files

bobc, 2017-09-21:
I have literally just encountered this one compiling for Due, the error is 
> fork/exec C:\Users\bob\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: The filename or extension is too long.

This is an unfortunate consequence of breaking Marlin up into more files with longer paths.

## Analysis
This problem is due to Windows having a limit of 32767 characters for a command, and Arduino IDEs final step is a link with a _very long_ list of object files. If this exceeds 32k you get the fork/exec error.

u8lib is a red herring, it just happens to be a library with quite a few files.

## Workarounds
Altering the path to the source files causes the length of the link command to vary, so that is one way to get under the limit.
Another way to shorten the command is to change your %TEMP% location to somewhere as short as possible, e.g. "C:\TMP".

fiveangle, 2017-09-21:
@thinkyhead - it looks like the new source tree is pushing many configs over the edge for Windows due to this issue.  I suspect we are going to have to come up with a blanket communication to all Windows users how to easily resolve this.

vMeph, 2017-09-21:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1322374/Configuration.txt)

fiveangle, 2017-09-21:
Yes, that is exactly true, and it does that because the new layout puts some files inside directories, and so the path length to the file has increased slightly.  This appears to be enough to exhaust the 32k command length limit in Windows.   It is not a Marlin issue, but an issue with Arduino IDE running on Windows.

There is nothing Marlin can do to resolve.  There are potential work-arounds suggested above.  If they do not help, I can only suggest to download something like VirtualBox and compile in Linux Arduino IDE. 

I am also investigating PlatformIO on Windows as a solution but that did not compile for your issue for another reason.

vMeph, 2017-09-21:
This just gonna be a problem for alot of people,hopefully someone comes up with a solution that be easy to work arround

Phr3d13, 2017-09-21:
I knew there was a reason I keep a linux box always running. :)

Phr3d13, 2017-09-21:
Do we know a rough number of characters we can get away with/vs how many is too many? if so, would a sanity check be able to check the number of characters in a particular file's full location and spit an error something like "Move/copy the Marlin directory to X location and try verify/upload again" if the actual characters exceeds the allowed characters?

fiveangle, 2017-09-21:
@Phr3d13, that would still be somewhat of a kludge as it may be that building Marlin at the root directory still may not fit (depending on the libraries used) so there would be no mechanism the user could move to a "shorter" path length.

@vMeph - Can you please open your issue at https://github.com/arduino/Arduino/issues ?  Perhaps they will have advice on how to work-around (perhaps an easy way to tell Arduino IDE to pass relative paths to the compiler on the cmdline.  Ideally though, Arduino IDE would detect the >32k command line and break up the linker into chunks in order to reduce the final link command to fit within the 32k limit.

Phr3d13, 2017-09-22:
Ok, as 2.0 rolls out, I'll keep my eyes on forums and groups that I'm in to try to help people with this issue before it gets spammed all over the issue tracker.

RalphSch, 2017-09-22:
I also ran into this issue and just found a way to change %TEMP%  with a litte batch file for the arduino.exe which sets environment variables for just one  application, see https://superuser.com/questions/1049430/how-do-you-set-environment-variables-for-a-single-command-on-windows. I have not yet tried it but I will test it this evening.
Even if this workaround helps it is no acceptable permanent solution because Windows/ArduinoIDE is the mostly used combination of the marlin users without any knowledge about variables, compilers and such nerdy stuff ;) Flashing Marlin is already very hard for them especially if it's compared to the smootieware solution if we look at the upcoming 32bit wave at 3D printing controllers. This workaround and also the platformio way would also make all the howtos, manuals and videos useless :\

bobc, 2017-09-22:
I think the ideal fix is for the Arduino builder to write the names of object files to a file, then use the `@file` option on gcc to include the list. I tested this with a batch file, and it seemed to work ok but I didn't include all 377 object files. The only wrinkle is that the path names of the object files must use forward slash and not the normal Windows backslash, but that is a simple conversion.

Form the user point of view, IF this fix was implemented in Arduino, then it would only mean using the latest Arduino IDE, which is the least impact I can think of.

In the meantime, the only workaround that doesn't require user action with paths etc is to reverse the breakup of gcode commands into seperate files, that has added about 130 new files to the link command which tips it over the brink.

vMeph, 2017-09-22:
ok i was able to fix by doing this

1- go to directory you want to create new temp folder and create a folder , in my case i called (Temp)
ex:  C:\arduino-1.8.4\Temp

2- open arduino and go to preferences and see where the preferences.txt file is saved
[image]
in this case is located in  C:\Users\vMeph\AppData\Local\Arduino15

3- go to that location and open preferences.txt file
on the 3th line add this line with the current path set on step 1
build.path=C:\arduino-1.8.4\Temp

 done

vMeph, 2017-09-22:
it seems eavean know i fixed the Marlin bug fix 2.0.x
not it seems to have a conflit with other versions of marlin
the error that comes now with marlin 1.1.x is
In file included from C:\arduino-1.8.4\Tem\sketch\src\inc/MarlinConfig.h:38:0,

                 from C:\arduino-1.8.4\Tem\sketch\src\Marlin.h:29,

                 from C:\arduino-1.8.4\Tem\sketch\src\Marlin.cpp:31:

SanityCheck.h:43: error: #error "You are using an old Configuration.h file, update it before building Marlin."

   #error "You are using an old Configuration.h file, update it before building Marlin."

    ^

SanityCheck.h:47: error: #error "You are using an old Configuration_adv.h file, update it before building Marlin."

   #error "You are using an old Configuration_adv.h file, update it before building Marlin."

    ^

exit status 1
#error "You are using an old Configuration.h file, update it before building Marlin."

Phr3d13, 2017-09-22:
if you tell me what example configs you use I can take a look. otherwise, just change the version string of the config files to match the rest.

fiveangle, 2017-09-22:
Honestly, you are far better off using a tool like BeyondCompare, WinMerge, etc, to compare the default Configuration*.h to your old ones, and hand-port your configuration to the new file.   If you just change the version you're on your own because if your configuration is complex (or very old) you will likely be plagued with a bunch of other failed sanity checks down the road because parts aren't in your config.

The version check is there for a reason.  Spend the 5 minutes to port it over rather than potentially hours tracking down why something "isn't working" IMO :)

GMagician, 2017-09-23:
**ANOTHER WORKAROUND:**

- from command prompt:
- type subst unit: marlingprojectpath (eg  subst r: c:\user\myname\Documents\Marlin)
- close prompt

Now from ide open project from unit (R:).
Unit is available till window shutdonw

thinkyhead, 2017-09-23:
It never occurred to me that such limitations would come into play. Is it possible, I wonder, to have the linker work incrementally, instead of linking everything in one go?

It's possible to compact these G-code handlers further, even to the point of having them grouped by feature. The original refactor concept was to compile all the G-code handlers under a single code unit, keeping them all in separate `.h` files, with the explicit understanding that all G-code handlers would be compiled under a single unit that includes them all. That kind of reorganization is still possible, simply be renaming all the g-code `.cpp` files as `.h` and then making a file that aggregates them all.

Rather than take that huge step backward, I'll take a stab at combining as many G-codes that belong together as possible, preserving the `Gnnn.cpp`/`Mnnn.cpp` file naming scheme.

p3p, 2017-09-23:
> It never occurred to me that such limitations would come into play

That would be because it really shouldn't, it's probably something that would never usually come up in Arduino IDE because of how badly it supports multiple files / folders in general.

bobc, 2017-09-23:
The Arduino builder creates a list of object files [ref 1](https://github.com/arduino/arduino-builder/blob/a07ea16effee9c7402f877c063ac1ac4702b856d/src/arduino.cc/builder/phases/linker.go#L51) which it puts into the `{object_files}` property, which is then used by the combine recipe [ref 2](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt#L68)

The exceptions to that are the core files, which are put into core.a, and any libraries which have the `dot_a_linkage` property  [Arduino library spec](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification). Also, in fact the some targets such as Due use an additional archive in the core (e.g. system.a) which is precompiled.

thinkyhead, 2017-09-23:
Too bad that Arduino has to compile in one stage and we can't define a custom build. Because then we could make the core of Marlin into a static library. It would add a bit of extra inconvenience. For config changes, you'd have to rebuild both the library and then the firmware.

GMagician, 2017-09-24:
I tryed to compile new marlin, after g-code grouping, seems not to solve!
My opinion is that if it's impossible to let arduino ide compile it's better to keep code layout readable and easy manageable

vMeph, 2017-09-24:
i agree with @GMagician 
if cant be solved would be better stay has it was :(

Roxy-3D, 2017-09-24:
> My opinion is that if it's impossible to let arduino ide compile it's better to keep code layout readable and easy manageable

> i agree with @GMagician   if cant be solved would be better stay has it was :(

This problem can be solved.   And we do need the code in a more organized and cleaner format.

fiveangle, 2017-09-25:
Arduino team already has proof-of-concept debug build that works around the issue.  Until a permanent solution is released, you can replace `arduinio-builder.exe `in your arduino installation with the debug one they provided here:

https://github.com/arduino/Arduino/issues/6751#issuecomment-331879017

fiveangle, 2017-09-25:
> @vMeph: 
> 
> > i agree with @GMagicianif cant be solved would be better stay has it was :(
> 

Fear of regression is never a reason to stagnate development.  2.0 isn't even released yet anyway, so this is where these things are expected to be resolved.

The failure is with Arduino, not Marlin and they now have pressure to fix it officially.   Until then, you can use the debug `arduino-builder.exe` binary linked above, PlatformIO instead of Arduino IDE, use a Linux VM with Arduino IDE on free VirtualBox emulator, and of course, there's always 1.1.5 if the 2.0 kitchen is too hot right now ;)

bobc, 2017-09-26:
The arduino-builder fix looks good and should work with any toolchain (I wonder which toolchains don't use a version of gcc, I guess there must be some).

I looked at the Due build with u8g lib, and there are 347 object files with average path length of object file names of about 39 bytes (excluding "c:\temp\" prefix). Add 1 byte for a space and 2 bytes for quotes, let's call that 40 bytes. That allows about 800 objects files, give or take. 800 files should be enough for everyone?

It's also worth noting that with the current method of using guards in HAL files, the object file still gets included in the link command even if the object file has no data. In the long term, with several HALs and supporting code, we might find we get close to the 800 file mark, we are currently about 350/800.

GMagician, 2017-10-02:
@fiveangle maybe I explained in a bad way or misunderstood your reply, I don't mean to return to 1.x.x layout but to keep 2.0 as previous.
I wouldn't limit 2.0 to history and lose readability just to keep compatibility with a "bugged/limited" tool.
avrdude or arduino-build (on your post) can be the solution until arduino ide will be fixed

thinkyhead, 2017-10-16:
So far I haven't run into issues building 2.0.x under Arduino IDE, but I don't use too many options. I'll mess around and see how many options it takes before things start to break. In the meantime, leave this issue open or continue discussion until the toolchain is fixed?

fiveangle, 2017-10-16:
It’s relevant for Windows only if course.  The promised  “31k command line length” build of arduino-builder.exe is still not available from https://github.com/arduino/Arduino/issues/6751#issuecomment-331879017  but I guess leaving this open won’t really do much since people rarely search until after opening their issue anyway :)

I’ve been linking people  directly to the arduino tracker issue rather than here. 

Adding a U8glib display will usually put it over the top.

Phr3d13, 2017-11-03:
Can this be closed?

closed 2018-08-12 by vMeph