MarlinFirmware/Marlin #7708: [FR] Add bed leveling status on LCD

MasterPIC, 2017-09-21:
What about adding a bed leveling status flag on the LCD?
It could be useful to be sure about the status of the (mesh) bed leveling...

Roxy-3D, 2017-09-21:
Agreed.   Perhaps you suggest the appropriate place and way to indicate the status on both the Graphical and 20x4 LCD screens?

Maybe a picture of each with the suggested modifications.

MasterPIC, 2017-09-21:
I would suggest a simple char on 20x4 LCD screen next to "Z" (e.g. "_") to be shown in case of active leveling.
For graphical screen I think there is space for further details.
Together with the added LCD flag, a message by LCD as soon as leveling is enabled /disabled would be another nice feature (also to overcome the limited space offered by 20x4 LCDs...).

fiveangle, 2017-09-21:
Great idea ! :1st_place_medal: 

-=dave

Phr3d13, 2017-09-21:
Would a progress bar/% complete be feasible? Filling up one char on a 20x4 lcd, more robust on a graphical lcd?

Roxy-3D, 2017-09-21:
% done on bed leveling doesn't really make sense.   The Mesh based leveling systems just load a mesh from EEPROM and they are 100% functional.

Phr3d13, 2017-09-21:
My bad, oddly, I have both, and use neither. Was gonna "upgrade" to them, but was already too used to the machines when they finally came in the mail.

fiveangle, 2017-09-21:
Honestly, the most useful wouldn't be progression, but visual indicator on if bed levelling is active or not (and works across all flavors).

Phr3d13, 2017-09-21:
I see that now, didn't completely have my head wrapped around it all when I made the first comment

Tannoo, 2017-09-22:
`About Printer` in the Info menu shows what bed leveling scheme is in use.

lrpirlet, 2017-09-22:
What about using lower case x, y and z when "compensated" coordinates are in use... Something as simple as that would allow to get the status at a glance instead of having to search for it into a specific screen...

Roxy-3D, 2017-09-22:
> What about using lower case x, y and z when "compensated" coordinates are in use... Something as simple as that would allow to get the status at a glance instead of having to search for it into a specific screen...

Or...  Maybe even an 'Extended Character' when the top bit is set.   Check out section 6 of: https://web.stanford.edu/class/ee281/handouts/lcd_tutorial.pdf    I think we could consider 0x1101000, or  0x11000110 or maybe even 0x10111100.

MasterPIC, 2017-09-23:
"About printer" doesn't tell if leveling is active. There is space for improvements here too.
Anyway I think it's worth having the state always shown (on the main screen).
Lower case obliges the user to remember the associated leveling state (ON or OFF?). Maybe a custom char (in place of Z) can be preferred.
Or, if adding a char next to Z is worth it, a ready to use underscore could be used (to report that Z coordinate is processed by the firmware to take care of leveling).

MasterPIC, 2017-09-25:
I just made a pull request (limited to HD44780 LCDs).

closed 2017-09-28 by MasterPIC