MarlinFirmware/Marlin #7710: [FR] Duel Heat Bed Support

Shadeling3D, 2017-09-22:
Building a printer driver board to run Marlin, I need support for 2 heat beds with Therm support please

Shadeling3D, 2017-10-12:
Under Pins.h:

#define BED_PINS HEATER_BED_PIN, marlinAnalogInputToDigitalPin(TEMP_BED_PIN)

would changing to the following help me? am I looking in the correct direction?
#define BED_PINS HEATER_BED0_PIN, marlinAnalogInputToDigitalPin(TEMP_BED0_PIN)
#define BED_PINS HEATER_BED1_PIN, marlinAnalogInputToDigitalPin(TEMP_BED1_PIN)

thinkyhead, 2017-11-15:
It would take a lot more than that to add support for 2 heated beds to Marlin. It is hard-coded to support only a single bed at this time.

Shadeling3D, 2017-11-21:
Thank you for the update, I'm leaving it as usable future expansion for now

sjasonsmith, 2020-11-10:
Duplicate of #5098

closed 2020-11-10 by sjasonsmith

sjasonsmith, 2020-11-29:
Duplicate of #4112