MarlinFirmware/Marlin #7712: Fan not respoding

Radiu, 2017-09-22:
hi using firmware Marlin-1.1.x
i have conected a fan on D9 on ramps 1.4

when i sent command M106 S250 fan started
now when send M107 nothing happens, if i try M106 S50 nothing happens same speed

can figuer out what is wrong, do i miss something?

thanks

Phr3d13, 2017-09-23:
Digging through [here](http://forums.reprap.org/read.php?219,393093,393093) a bit hints at hardware inconsistencies. You could try following the suggested diagnostic measures if applicable.

Tannoo, 2017-09-23:
Try this first..

Tannoo, 2017-09-23:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

closed 2017-09-23 by Radiu