MarlinFirmware/Marlin #7714: hotend E3D-V6 dont stop heating

vMeph, 2017-09-22:
Hi im trying set up my hotend E3D-V6 but soon i start machine it starts always heating up
i cant seem to see what is wrong with my configuration files
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1326013/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1326015/Configuration_adv.txt)
[pins_RAMPS.txt](https://github.com/MarlinFirmware/Marlin/files/1326016/pins_RAMPS.txt)

the hot end is pluged on D10
the temp 0 is pluges on T0

not sure what could be wrong any help would be apreciated
thanks

Phr3d13, 2017-09-23:
Check that you don't have a short between +3.3V and ground, because that will turn the hot end and bed heaters on. The most common way of shorting +3.3V to ground is to wire the endstop switches incorrectly.

vMeph, 2017-09-23:
Ok found problem was a bad solder on the mosfet, now is working,lol
Thanks alot

closed 2017-09-23 by vMeph