MarlinFirmware/Marlin #7716: Marlin 111 z slams bed after home/bltouch leveling& start print

allcustomz, 2017-09-23:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.
Need help z axis slams into bed after auto home, bl-touch leveling, start print??
------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

allcustomz, 2017-09-23:
Kinda lost on wgat to do any help would be really appreciated thanks love these forums they gave helped me alot hopefully someone eles has same issue and we can all get running again

p3p, 2017-09-23:
You need to either edit this issue following the instructions rather than just posting them then someone may be able to help, for a start I don't think this is a feature request. (follow the bug report section then remove the rest)

allcustomz, 2017-09-23:
Oh ok im on my phone so i cant add all errors sorry about that i just delete

closed 2017-09-23 by allcustomz