MarlinFirmware/Marlin #7718: ABL_BILINEAR_SUBDIVISION

nygellyndley, 2017-09-23:
I'm having an issue with AUTO_BED_LEVELING_BILINEAR with ABL_BILINEAR_SUBDIVISION enabled.  I'm not sure at this time is it is a bug but it seems like one.

Steps to reproduce :

- enable AUTO_BED_LEVELING_BILINEAR
- set GRID_MAX_POINTS_X/Y to 7
- enable ABL_BILINEAR_SUBDIVISION
- set BILINEAR_SUBDIVISIONS to 3
- G28
- G29
- G0 X0 Y0 Z20 F2000

The extruder moves from it's finishing G29 position diagonally to X0 Y0, then once then goes down right into the print bed.

A normal G0 X0 Y0 Z20 F2000 goes diagonally X,Y,Z in one move to the correct position 0,0,20 rather than X,Y then (incorrect) Z.

Thinking it might be some out of memory issue, I've tried dropping grid sizes to 5 and subdivisions to 1 but to no avail.

If I disabled the bilinear subdivisions it all seems fine.

Additionally I also see blank spots in the Catmull (=======), though this may be normal I'm not sure, it just seems odd, ie :

`================================================

Recv:       0      1      2      3      4      5      6
Recv:  0 -0.863 -0.640 -0.440 -0.227 -0.063 +0.100 +0.263
Recv:  1 -0.730 -0.550 -0.365 -0.145 +0.030 +0.175 +0.335
Recv:  2 -0.645 -0.445 -0.267 -0.060 +0.095 +0.250 +0.410
Recv:  3 -0.590 -0.387 -0.222 -0.022 +0.173 +0.310 +0.460
Recv:  4 -0.525 -0.317 -0.150 +0.040 +0.210 +0.368 +0.497
Recv:  5 -0.445 -0.275 -0.122 +0.093 +0.255 +0.413 +0.538
Recv:  6 -0.385 -0.212 -0.037 +0.165 +0.328 +0.455 +0.648
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6
Recv:  0 -0.86250 -0.64000 -0.44000 -0.22750 -0.06250 +0.10000 +0.26250
Recv:  1 -0.73000 -0.55000 -0.36500 -0.14500 +0.03000 +0.17500 +0.33500
Recv:  2 -0.64500 -0.44500 -0.26750 -0.06000 +0.09500 +0.25000  =======
Recv:  3 -0.59000 -0.38750 -0.22250 -0.02250 +0.17250 +0.31000  =======
Recv:  4 -0.52500 -0.31750 -0.15000 +0.04000 +0.21000 +0.36750  =======
Recv:  5 -0.44500 -0.27500 -0.12250 +0.09250 +0.25500 +0.41250  =======
Recv:  6 -0.38500 -0.21250 -0.03750 +0.16500 +0.32750 +0.45500  =======

================================================

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5
Recv:  0 -0.795 -0.575 -0.345 -0.135 +0.050 +0.210
Recv:  1 -0.655 -0.452 -0.255 -0.052 +0.130 +0.300
Recv:  2 -0.575 -0.367 -0.157 +0.065 +0.225 +0.388
Recv:  3 -0.490 -0.280 -0.102 +0.108 +0.290 +0.428
Recv:  4 -0.412 -0.232 -0.030 +0.168 +0.340 +0.497
Recv:  5 -0.335 -0.140 +0.053 +0.240 +0.405 +0.565
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5
Recv:  0 -0.79500 -0.57500 -0.34500 -0.13500 +0.05000 +0.21000
Recv:  1 -0.65500 -0.45250 -0.25500 -0.05250 +0.13000 +0.30000
Recv:  2 -0.57500 -0.36750 -0.15750 +0.06500 +0.22500 +0.38750
Recv:  3 -0.49000 -0.28000 -0.10250 +0.10750 +0.29000  =======
Recv:  4 -0.41250 -0.23250 -0.03000 +0.16750 +0.34000 +0.49750
Recv:  5 -0.33500 -0.14000 +0.05250 +0.24000 +0.40500  =======

`

Roxy-3D, 2017-09-23:
I think you are on to something.

nygellyndley, 2017-09-24:
Just tried 1.4 (with what I believe is an equivalent config
[Configuration1.4.zip](https://github.com/MarlinFirmware/Marlin/files/1327100/Configuration1.4.zip)
) to see, and the issue does not seem to occur there.

Using AUTO_BED_LEVELING_BILINEAR with ABL_BILINEAR_SUBDIVISION enabled, the following `G0 X0 Y0 Z20 F2000` after the `G29` seems to behave normally.

And the CatMull data looks fine, though again I have no idea if the above CatMull is incorrect, I am new to 3d printing :

```Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6
Recv:  0 -0.810 -0.625 -0.445 -0.288 -0.103 -0.000 +0.150
Recv:  1 -0.745 -0.525 -0.408 -0.165 -0.030 +0.117 +0.240
Recv:  2 -0.640 -0.438 -0.283 -0.120 -0.010 +0.147 +0.272
Recv:  3 -0.595 -0.383 -0.233 -0.088 +0.095 +0.225 +0.322
Recv:  4 -0.528 -0.320 -0.163 +0.010 +0.147 +0.245 +0.367
Recv:  5 -0.468 -0.283 -0.158 +0.022 +0.155 +0.320 +0.440
Recv:  6 -0.405 -0.268 -0.108 +0.067 +0.232 +0.370 +0.440
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18
Recv:  0 -0.81000 -0.74815 -0.68630 -0.62500 -0.56380 -0.50315 -0.44500 -0.39185 -0.34120 -0.28750 -0.22481 -0.15907 -0.10250 -0.06398 -0.03463 -0.00000 +0.04648 +0.09824 +0.15000
Recv:  1 -0.78981 -0.72201 -0.65421 -0.59120 -0.53754 -0.48867 -0.43574 -0.37355 -0.30731 -0.24380 -0.18438 -0.12768 -0.07639 -0.03396 +0.00306 +0.04241 +0.08743 +0.13478 +0.18213
Recv:  2 -0.76963 -0.69588 -0.62213 -0.55741 -0.51128 -0.47418 -0.42648 -0.35525 -0.27342 -0.20009 -0.14394 -0.09629 -0.05028 -0.00395 +0.04075 +0.08481 +0.12838 +0.17132 +0.21426
Recv:  3 -0.74500 -0.66787 -0.59074 -0.52500 -0.48287 -0.45213 -0.40750 -0.33194 -0.24250 -0.16500 -0.11250 -0.07194 -0.03000 +0.01917 +0.06972 +0.11750 +0.16019 +0.20009 +0.24000
Recv:  4 -0.71074 -0.63496 -0.55917 -0.49370 -0.44920 -0.41501 -0.36954 -0.29794 -0.21506 -0.14380 -0.09690 -0.06161 -0.02259 +0.02773 +0.08179 +0.13222 +0.17538 +0.21491 +0.25444
Recv:  5 -0.67204 -0.60017 -0.52830 -0.46380 -0.41337 -0.37032 -0.32185 -0.25893 -0.19060 -0.13120 -0.09029 -0.05833 -0.02102 +0.02939 +0.08514 +0.13722 +0.18142 +0.22196 +0.26250
Recv:  6 -0.64000 -0.57074 -0.50148 -0.43750 -0.38259 -0.33296 -0.28250 -0.22694 -0.17056 -0.12000 -0.08120 -0.04824 -0.01000 +0.04019 +0.09565 +0.14750 +0.19157 +0.23204 +0.27250
Recv:  7 -0.62139 -0.55148 -0.48157 -0.41704 -0.36124 -0.31082 -0.26102 -0.20977 -0.15915 -0.11065 -0.06621 -0.02389 +0.02065 +0.07090 +0.12338 +0.17194 +0.21325 +0.25065 +0.28806
Recv:  8 -0.60944 -0.53758 -0.46572 -0.40019 -0.34493 -0.29600 -0.24787 -0.19962 -0.15217 -0.10269 -0.04873 +0.00726 +0.06074 +0.11075 +0.15827 +0.20167 +0.23904 +0.27230 +0.30556
Recv:  9 -0.59500 -0.52185 -0.44870 -0.38250 -0.32769 -0.27981 -0.23250 -0.18519 -0.13843 -0.08750 -0.02750 +0.03667 +0.09500 +0.14343 +0.18602 +0.22500 +0.25991 +0.29120 +0.32250
Recv: 10 -0.57389 -0.50082 -0.42775 -0.36130 -0.30581 -0.25694 -0.20824 -0.15846 -0.10885 -0.05667 +0.00204 +0.06332 +0.11806 +0.16151 +0.19841 +0.23389 +0.26916 +0.30301 +0.33685
Recv: 11 -0.55028 -0.47796 -0.40563 -0.33926 -0.28300 -0.23270 -0.18176 -0.12745 -0.07252 -0.01861 +0.03533 +0.08823 +0.13528 +0.17240 +0.20367 +0.23639 +0.27350 +0.31207 +0.35065
Recv: 12 -0.52750 -0.45648 -0.38546 -0.32000 -0.26435 -0.21426 -0.16250 -0.10481 -0.04546 +0.01000 +0.05991 +0.10593 +0.14750 +0.18204 +0.21213 +0.24500 +0.28398 +0.32574 +0.36750
Recv: 13 -0.50704 -0.43759 -0.36814 -0.30481 -0.25280 -0.20690 -0.15769 -0.09923 -0.03746 +0.01926 +0.06702 +0.10974 +0.15074 +0.19024 +0.22802 +0.26685 +0.30798 +0.35015 +0.39231
Recv: 14 -0.48741 -0.42009 -0.35276 -0.29241 -0.24539 -0.20535 -0.16009 -0.10202 -0.03873 +0.01907 +0.06544 +0.10632 +0.14898 +0.19717 +0.24713 +0.29481 +0.33812 +0.37915 +0.42019
Recv: 15 -0.46750 -0.40361 -0.33972 -0.28250 -0.23843 -0.20102 -0.15750 -0.09981 -0.03602 +0.02250 +0.06898 +0.11019 +0.15500 +0.20926 +0.26713 +0.32000 +0.36333 +0.40167 +0.44000
Recv: 16 -0.44685 -0.38839 -0.32992 -0.27583 -0.23079 -0.19012 -0.14417 -0.08651 -0.02356 +0.03509 +0.08398 +0.12858 +0.17565 +0.23036 +0.28755 +0.33852 +0.37829 +0.41183 +0.44537
Recv: 17 -0.42593 -0.37419 -0.32246 -0.27167 -0.22359 -0.17645 -0.12583 -0.06821 -0.00710 +0.05130 +0.10412 +0.15424 +0.20407 +0.25662 +0.30887 +0.35426 +0.38831 +0.41550 +0.44269
Recv: 18 -0.40500 -0.36000 -0.31500 -0.26750 -0.21639 -0.16278 -0.10750 -0.04991 +0.00935 +0.06750 +0.12426 +0.17991 +0.23250 +0.28287 +0.33019 +0.37000 +0.39833 +0.41917 +0.44000```

So this is looking like an issue introduced in 1.5?

empakoso, 2017-09-26:
I just had my config (with UBL active)drive into the bed under the same steps as mentioned this past weekend. Been running 1.1.5 since the beginning of Sept when it got released, never until now.  Didn’t matter what G1 command I gave it; G1 X100 Y105, or G1 Z40, etc My nozzle just bore down into my bed. Wouldn’t stop. I had to load my V1.1.5 and jettisoned all my meshes to get it to work again.
I will post my mesh and my configs later.

Roxy-3D, 2017-09-26:
@empakoso   What type of probe do you have?   Is it a BL-Touch type of Z-Probe?

>  Been running 1.1.5 since the beginning of Sept when it got released
>  I had to load my V1.1.5 and jettisoned all my meshes to get it to work again.

You've been running v1.1.5 but had to move back to v1.1.5 ?  That doesn't make sense.   And you can preserve your mesh data by doing a G29 S-1.   It won't matter if the EEPROM changes...  You can re-load the mesh data and save it.

empakoso, 2017-09-26:
Stationary IR probe from filastruder.  Kept driving into my bed so reloaded the same config. You are right I reloaded my saved mesh using G29 S-1 and saved to S0. I kept/keep a backup in slot 9.

Roxy-3D, 2017-09-26:
UBL uses the same lower level probing system that the other bed leveling systems use.   It might be good to switch to another system and see if you still have the issue.   (I think you will.)     Knowing the answer to that question would help us figure out where to look for the problem.

empakoso, 2017-09-26:
Ok I will switch over and respond back.

empakoso, 2017-10-01:
I have not let this go.  I am extremely busy but have every intention to switch to another bed levelling system soon.  I will stay on this.

empakoso, 2017-10-06:
Here is the latest update.
I have not yet changed to BiLinear mesh.
My UBL mesh shifting down by approx 0.2 so did a fine tuning of the bed.
Noticed strange movements when I started a print;
Each z lift of 0.25
It would climb higher and higher
The zlift would not return down
The lcd would display the layer height it was printing on and wouldn’t change as the nozzle kept rising on every lift.

Roxy-3D, 2017-10-06:
What version (date) of Marlin are you using?   There used to be an issue similar to that with regard to the firmware retract.    Do you have firmware retract turned on?   If you are not using the current version of the code, can you load it?

empakoso, 2017-10-06:
Downloaded bug-fix on sept.3
Yes fwRetract is enabled.
Ok. Will download the latest and get back.
This weekend is our long weekend so should have some more time.

Roxy-3D, 2017-10-06:
If you see the problem with the latest download...  Please turn off firmware retract.   And please tell us that the Z-Hop problem is still not fixed!

empakoso, 2017-10-09:
The latest download has fixed the issue.

closed 2018-02-09 by thinkyhead