MarlinFirmware/Marlin #7724: Understanding Heater Settings Error; temperature.cpp

kcheeeung, 2017-09-24:
Hello everyone,

I would like to understand the way the Marlin temperature are written and why the way it is written. For my 3D printing application, I am heating a syringe heater. During my test run, I enter the command M104 S100. Everything is normal as expected. Next, I enter M104 S150.

A bit later I receive the message and calls a kill to stop the entire print.
**Error: Heating failed, system stopped Heater_ID:0**
```
[void Temperature::_temp_error(const int8_t e, const char * const serial_msg, const char * const lcd_msg) {
  static bool killed = false;
  if (IsRunning()) {
    SERIAL_ERROR_START();
    serialprintPGM(serial_msg);
    SERIAL_ERRORPGM(MSG_STOPPED_HEATER);
    if (e >= 0) SERIAL_ERRORLN((int)e); else SERIAL_ERRORLNPGM(MSG_HEATER_BED);
}]
```
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/temperature.cpp#L509

```
// Extruder temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.

// Bed temperature must be close to target for this long before M190 returns success
#define TEMP_BED_RESIDENCY_TIME 10  // (seconds)
#define TEMP_BED_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_BED_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.
```

What is the underlying mechanism of what causes this error to trigger? I know there are several fire protection and safety mechanisms in the Marlin firmware, but unsure of why this happens. Does the config settings have to do with anything?  I have provided the setup of my heater and board design if that matters.

[image]

AnHardt, 2017-09-25:
Maybe reading the relevant part in Configuration_adv.h can help you.

```
/**
 * Thermal Protection protects your printer from damage and fire if a
 * thermistor falls out or temperature sensors fail in any way.
 *
 * The issue: If a thermistor falls out or a temperature sensor fails,
 * Marlin can no longer sense the actual temperature. Since a disconnected
 * thermistor reads as a low temperature, the firmware will keep the heater on.
 *
 * The solution: Once the temperature reaches the target, start observing.
 * If the temperature stays too far below the target (hysteresis) for too long (period),
 * the firmware will halt the machine as a safety precaution.
 *
 * If you get false positives for "Thermal Runaway" increase THERMAL_PROTECTION_HYSTERESIS and/or THERMAL_PROTECTION_PERIOD
 */
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

  /**
   * Whenever an M104 or M109 increases the target temperature the firmware will wait for the
   * WATCH_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M104/M109,
   * but only if the current temperature is far enough below the target for a reliable test.
   *
   * If you get false positives for "Heating failed" increase WATCH_TEMP_PERIOD and/or decrease WATCH_TEMP_INCREASE
   * WATCH_TEMP_INCREASE should not be below 2.
   */
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
#endif

```

kcheeeung, 2017-09-25:
Thanks so much! I think my heater design is part of the reason, and it might have been heating up too slowly, thus causing the Marlin to think the thermistor is out of place or failed. I will update these settings and improve my design.

closed 2017-09-25 by kcheeeung