MarlinFirmware/Marlin #7729: Usage question: UBL or Bilinear

batata004, 2017-09-24:
Hi,

As I see from the documentation UBL seens very similar to Bilinear mesh leveling. Currently I am using 4 points in the GRID with Bilinear. Should I expect any improvement using UBL or they are the same thing?

Roxy-3D, 2017-09-24:
If you have the mesh well tuned...  Both are going to give very similar results.   Mostly, UBL was designed to be full tool set to have the good features from all the previous bed leveling systems.  It also has an interactive mesh editor that can use the G26 Mesh Validation Pattern to help you fine tune the mesh.  Bi-linear is more focused on keeping the code size of the bed leveling system very small.

batata004, 2017-09-24:
@Roxy-3D thanks my friend. What do you mean by "mesh well tuned"? Do I have to tune my mesh bed leveling? I use a fixed mount using inductive sensor.

Roxy-3D, 2017-09-24:
> Do I have to tune my mesh bed leveling? I use a fixed mount using inductive sensor.

You tell me!   If you have a machine with good mechanics and an accurate probe...  You may not need to adjust the mesh after it is sampled.    But my experience is portions of the mesh are usually off a little bit.   That is the purpose of the G26 Mesh Validation Pattern.  It will draw a pattern on the bed so you can **_SEE_** how well each mesh point reflects reality.  Check out the end of this thread: 
 https://github.com/MarlinFirmware/Marlin/issues/7618#issuecomment-328297556

And then using that Mesh Validation Pattern, there is an interactive editor built into UBL  (G29 P4 R) that lets you edit areas of the mesh.   You can do the editing (if necessary) with M421, but UBL's interactive editor is easier and quicker.

batata004, 2017-09-24:
Wowo great!!! Thansk my friend that's a good idea printing that sample file and check how the result is gonna be! Good anwer. I thank you a lot!

closed 2017-09-24 by batata004

Roxy-3D, 2017-09-24:
If you have a well tuned mesh....   You should get good adhesion across 100% of the bed 100% of the time.