MarlinFirmware/Marlin #7730: Marlin 2.0 autolevel and dondolo

alexxy, 2017-09-24:
I cannot compile current 2.0.x branch with autolevel + dondolo with PlatformIO

`
Compiling .pioenvs/megaatmega2560/src/src/module/temperature.o
Compiling .pioenvs/megaatmega2560/src/src/module/tool_change.o
Compiling .pioenvs/megaatmega2560/src/src/sd/Sd2Card.o
Marlin/src/module/tool_change.cpp: In function 'void tool_change(uint8_t, float, bool)':
Marlin/src/module/tool_change.cpp:465:38: error: 'leveling_is_active' was not declared in this scope
if (leveling_is_active()) {
^
*** [.pioenvs/megaatmega2560/src/src/module/tool_change.o] Error 1

`

GMagician, 2017-09-25:
Can you attach your configurations files?
Till yesterday I was able to compile a similar H/W configuration...

alexxy, 2017-09-25:
Sure
[marlin_dondolo_mbl.zip](https://github.com/MarlinFirmware/Marlin/files/1330307/marlin_dondolo_mbl.zip)

GMagician, 2017-09-25:
Added pull request

alexxy, 2017-09-29:
Its fixed now in bugfix-2.0.x

closed 2017-09-29 by alexxy