MarlinFirmware/Marlin #7731: case light brightness

vMeph, 2017-09-24:
Marlin-1.1.x
case light brightness lcd control 
if light brightness is set at 255 it will work the on off button
if light brightness is less then 255 the on off button dont works

AnHardt, 2017-09-25:
Be sure to use hardware pwm-able pin.

vMeph, 2017-09-25:
Its o d4 it dimms perfect using lcd meno brigthness just dont works the on and off if is less the 255

Tannoo, 2017-09-25:
Driving the case light via a relay can also cause this.

vMeph, 2017-09-25:
Im using ULN2003 IC to drive

RalphSch, 2017-09-28:
If I understand vMeph right than he doesn't have a problem with dimming the case light. I saw this bug also, if you are using a pwm value lower then 255 then the lcd menu action on/off does not work at all.

vMeph, 2017-09-30:
ya thats what i tought, if ou put a value below 255 then dont works

jmdearras, 2017-10-25:
I have the same issue, using pin 2  anything under 255  is off.  Fet drive, no relay.

closed 2017-11-14 by vMeph

RalphSch, 2017-11-15:
Is it fixed?

jmdearras, 2017-11-15:
Yes, in latest bugfix 1.1.x