MarlinFirmware/Marlin #7732: [FR] Nozzle touch auto bed-leveling

vido89, 2017-09-24:
Add feature to marlin which will enable us to do accurate bed leveling by nozzle touching heated bed we can use slow speeds and even decrease speed to even slower when nozzle come 2cm near heated bed

Roxy-3D, 2017-09-24:
There are already ways to do manual probing.   More of an explanation would be helpful.

vido89, 2017-09-24:
Tnx for quick reply but explanation of what. Many people need this feature it should be implemented just let people decide do they gone use it or not

AnHardt, 2017-09-25:
Recent versions of Marlin already have all you need to probe automatically with the nozzle - including approximation speed and cleaning it before.

Roxy-3D, 2017-09-25:
> Tnx for quick reply but explanation of what. Many people need this feature it should be implemented just let people decide do they gone use it or not

Manual probing exists in several forms.  But if you turn on UBL...   You can manually probe the unreachable parts of your bed with G29 P2.    Please check out what is already in place.

vido89, 2017-09-25:
Thats great thank you very much cheers

vido89, 2017-09-25:
@Roxy-3D You say manual Im trying to do auto-bed leveling

Roxy-3D, 2017-09-25:
> You say manual I'm trying to do auto-bed leveling

All of the various Auto Bed Leveling schemes work best if the bed is physically leveled as much as possible.    Please check out https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/ubl_G29.cpp#L163-L187 to see how UBL's MANUAL PROBING works.

deram, 2017-09-26:
I'd like to use UBL and its G29 P2 B, but UBL won't compile without a probe, and PROBE_MANUALLY is not counted as one anymore.

vido89, 2017-09-26:
@Roxy-3D Thank you also **[this](http://marlinfw.org/docs/features/unified_bed_leveling.html)** helped a lot

RalphSch, 2017-09-28:
I'm using the direct nozzle touch method now for some weeks and it works great. There are some problems, like homing with cold filament .... take a look at #7680 ;)

closed 2018-02-09 by thinkyhead