MarlinFirmware/Marlin #7739: Ultimaker 2 LCD Panel + Marlin

kr15uk, 2017-09-25:
Can I use Ultimaker 2 LCD with Marlin + RAMPS like board?! Tried uncomment Ultipanel but that seems to be Ultimaker original panel config.

I'm retrofitting my Ultimaker 2 with RAMPS like board so I have more control and expansion capabilities but want to keep inbuilt LCD (I don't want Ultimaker menu system, just regular Marlin please interface)

Thanks.

Kris

dot-bob, 2017-09-26:
The Ultimaker 2 display uses I2C as it's communication interface.  You will need to wire up these signals as well as the encoder signals to gpio's and SPI for the SD card.  While I don't have a Ultimaker 2 I do have a Duplicator 6 which is almost an exact copy from an electronics standpoint.  If you want to see my configuration for the Duplicator 6 it is under my MMU branch here: [https://github.com/dot-bob/Marlin/tree/MMU](url).  Mine uses a BLTouch probe instead of the regular Z-endstop like the Ultimaker 2.

boelle, 2019-02-19:
@kr15uk did you try as @dot-bob said?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

OP have not replied back and @dot-bob did suggest how it can be connected

thisiskeithb, 2022-02-26:
Marlin supports the Ultimaker 2 and its `ULTI_CONTROLLER` LCD, so you just need to wire it properly for your board if it supports i2c.

closed 2022-02-26 by thisiskeithb