MarlinFirmware/Marlin #7744: Bug with CUSTOM_USER_MENUS

batata004, 2017-09-25:
Hi,

In previous version of marlin (1.1.3) this code worked perfectly:

```
#if ENABLED(CUSTOM_USER_MENUS)
  #define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK

  #define USER_DESC_1 "First Script"
  #define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND) "\nG28\nG29"
....
```

I mean: when I click the FIRST SCRIPT option the menu would beep and right after that it would come back to the main screen (Where shows temperature and other stuff).

In the newest version (1.1.5) when I click FIRST SCRIPT option (with the knob) marlin keeps (ok!) and starts doing the job BUT the screen remains the same giving the impression something went wrong! I think marlin should come back to the main screen so the user can see what is being done (in the bottom line).

Roxy-3D, 2017-09-25:
@Tannoo     I bet this is the same thing that is messing up the Reselect of the Same file for SD prints.

Tannoo, 2017-09-25:
I don't think it's the same thing. Because what I found was only an issue when led printer events was enabled.

batata004, 2017-09-25:
I sincerally have no idea why this bug is happening but in version 1.1.3 it does not happen.

Tannoo, 2017-09-26:
Go back to the main screen and see if the status line says "User script done".

Also, can open a serial monitor to see if anything was done when you clicked.

Tannoo, 2017-09-26:
Did you change the first script from the default script?

batata004, 2017-09-26:
@Tannoo my configuration_adv is like this in the part you asked:

```
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  #define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK

  #define USER_DESC_1 "First Script"
  #define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND) "\nG28\nG29"

  //#define USER_DESC_2 "Preheat for PLA"
  //#define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  //#define USER_DESC_3 "Preheat for ABS"
  //#define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  //#define USER_DESC_4 "Heat Bed/Home/Level"
  //#define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

  //#define USER_DESC_5 "Home & Info"
  //#define USER_GCODE_5 "G28\nM503"
#endif
```

It only shows User script done when the "USER_GCODE_1" is executed, which happens after about 5 minutes.

I just tried this bug with version 1.1.2 and it also does not happen there. I had it here in my computer and tried it, so the bug does not happen in version 1.1.2 nor 1.1.3.

Tannoo, 2017-09-26:
Ok. I am trying it with `bugfix-2.0.x`.

Tannoo, 2017-09-26:
Add this to `Configuration_adv.h`:

```
  // Enable USER_SCRIPT_RETURN to return to the status screen after script is complete
  #define USER_SCRIPT_RETURN
```

And change this in `ultralcd.cpp`:

```
    void _lcd_user_gcode(const char * const cmd) {
      enqueue_and_echo_commands_P(cmd);
      #if ENABLED(USER_SCRIPT_AUDIBLE_FEEDBACK)
        lcd_completion_feedback();
      #endif
    }
```

to:

```
    void _lcd_user_gcode(const char * const cmd) {
      enqueue_and_echo_commands_P(cmd);
      #if ENABLED(USER_SCRIPT_AUDIBLE_FEEDBACK)
        lcd_completion_feedback();
      #endif
      #if ENABLED(USER_SCRIPT_RETURN)
        lcd_return_to_status();
      #endif
    }
```

See if that helps. This gives the option to return to the status screen after every user script.

Tannoo, 2017-09-26:
I remember someone asking to not return to the status screen as they wanted to run multiple scripts.

Tannoo, 2017-09-26:
The original PR (#6895) did return to status:

```
    void _lcd_user_gcode(const char * const cmd) { 
      lcd_return_to_status(); 
      enqueue_and_echo_commands_P(cmd); 
   } 
```

batata004, 2017-09-26:
@Tannoo  THAT WORKED PERFECTLY!!!!!! Hope this code will be added to the future versions of Marlin in the main branch! Thanks a lot, it worked like a charm!

closed 2017-09-26 by batata004

Tannoo, 2017-09-27:
@Roxy-3D, if I were to create a PR, what branch should I make it against?

Roxy-3D, 2017-09-27:
I think bugfix-2.0.x     That is where all activity is going to merge back together.   The big question is this:  Can you build it and make it work on the AVR (Atmega's) with your changes?    If the answer is "Yes." then for sure that is the right answer.

Tannoo, 2017-09-27:
Yes... that is what I'm ensuring still works.

Tannoo, 2017-09-27:
I haven't messed with the Re-ARM in a while.

thinkyhead, 2017-09-27:
The option has been merged with 1.1.x and 2.0.x.

batata004, 2017-09-27:
Great but I didnt understand one thing... when Marlin gets updated to the version 1.1.6 (for example), will it contain this fix that you guys provided?

Tannoo, 2017-09-27:
It will now.

Tannoo, 2017-09-27:
#7765 was my PR for this.