MarlinFirmware/Marlin #7745: Wrong x carriage moved when active tool = T1 for dual_x_carriage configuration

w1ebr, 2017-09-25:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: --- Move x axis does not account for current tool in dual_x_carriage configuration
- Expected behaviour: --- After selecting T1, an x axis move should move the second x carriage. 
- Actual behaviour: --- The carriage x location as shown on the LCD is correct, but a move x axis only moves the T0 carriage, regardless of the active tool.
- Steps to reproduce:
  - Set the active tool to T1 (done via a custom menu)
  - Move the carriage on the x axis.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1329936/Configuration.zip)

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2017-09-27:
Ah, good point… I'll make a patch for this today.

w1ebr, 2017-10-01:
Was the patch made?
Thanks!

thinkyhead, 2017-10-01:
Working on this at the moment. I hope to have a patch soon.

thinkyhead, 2017-10-03:
Patched! Turned out to be a simple solution.

closed 2017-10-03 by thinkyhead

w1ebr, 2017-10-03:
NOOB question: How do I pick up this patch?

w1ebr, 2017-10-12:
This does not appear to be fixed in 1.1.6. Is there a patch to 1.1.6 that I should make to get this?
Thanks!