MarlinFirmware/Marlin #7754: [FR] Adding Nextion HMI Display support

DaeMonSx, 2017-09-26:
How about to add the Nextion display support as a standard lcd feature?
I do know, there is the marlinkimbra with that feature, but would be good to have it in the main line.

fiveangle, 2017-09-27:
@apballard has done an absolutely gorgeous implementation:

https://www.youtube.com/watch?v=4Nn
http://support.iteadstudio.com/support/discussions/topics/11000008599
https://github.com/apballard/Marlin

-=dave

Tannoo, 2017-09-28:
I was working on that also for a little while and ran into the snag of adding screens. It snowballed into something I gave up on for the time being.

Deneteus, 2017-09-28:
And my axe!  Actually, I would like to see some sort of external serial display functionality so I can do stuff like add an Android display as an external monitor for multiple printers.

apballard, 2017-09-29:
@fiveangle i want to re-do my implementation for version 2. Want the development to stabilize a bit and maybe have v2 released.

#7790 could also be addressed at the same time. We should look at doing a serial lcd framework that has a default protocol defined. That would mean a second serial port, with the ability to receive gcode and pass it over to normal handlers. And some more specific lower level functions for setting temperature and so forth much like the current lcd implementation.

Would have to re-write the nextion library into a more generic library/driver so it can be used by other serial displays.

fiveangle, 2017-09-29:
@Deneteus — Oh, all we need to do is submit a feature request ? Oh wow, where do I begin ! ;)

@apballard — your implementation has quite remarkable polish for a first pass proof of concept. Nice job. I'm sure if you ever find time to get it to v0.82 status you'll have more people willing to test out stuff for you that you would know what to do with. 

Let us know ahead of time so we can each get one ordered and put on the slow boat to us for when you're ready :)

Deneteus, 2017-09-29:
@fiveangle  MagoKimbra solved this in 2015. It's a little beyond a feature request at this point. 

[image]

lkvapil, 2017-12-06:
Any news @Deneteus @fiveangle  ?

Spawn32, 2018-02-15:
Any news on this ?  😄

lkvapil, 2018-02-15:
I'm too interested :D

weeelfi, 2018-02-20:
Hi!

I test the @apballard solution with Mega+Ramps, and works perfect. (only need to fix a some parentheses missing, nothing more)

I think that another kind of LCD is a good feature for Marlin. Nextion is more expensive than RepRapDiscount Full Graphic (for example) but is easier to use and more intuitive (tactile > knob)

alexxy, 2018-06-20:
Any news on this type of displays?

thinkyhead, 2018-06-21:
Nothing to report at this time.

BigDi, 2018-08-30:
Tried to merge the HMI part of apballard to Marlin 2.0. I have a LPC1768 Board and copiling stops in ITEADLIB Arduino library at a function that will set a timeout on the serial. I also made some update in the Arduino code. But here ends my knowledge...

aenertia, 2018-09-03:
FYI - a lot of the STM32 based Printers are using Nextion HMI displays I have one sitting in front of me right now from my Mooz2. 

I can't say I am a huge fan of them, the nextion editor is pretty closed up; there are some python libraries that can sorta update the firmware over serial - but I couldn't get them to work (probably due to the current firmwares often locking them out). 

It will likely mean defining a standard 'Marlin' UI and shipping a compiled tft blob to load on them for the various size variants AFAIK there doesn't appear to be a Opensource build chain. So I would guess it will mean defining standard elements for the generic case and then having someone implement them on the various screen sizes /feature touch/non touch. 

TBH the cheap knockoff waveshare xpt2046 - sitting next to it which whilst limited to extremely slow framebuffer support is actually much more useful.

However given the prevalence of the nextion screens appearing in printer kits - it's probably worthwhile to have a plan around how to manage this.

BigDi, 2018-09-04:
I have no knowledge about connection between display to Marlin. I was looking for a color touch display for my DYI printer. I changed last weeks from RAMPS to SBASE. I ordered a TFT28 but I'm not satisfied. I found this post. Before I order a Nextion I want to try the code from apballard. But without sucess. I search for other possibilties and found also display from AMP. They used same methods like Nextion. But they discribe the binary [protocol](http://www.ampdisplay.com/documents/pdf/DGUS%20v2.0%20(08-09-12.pdf). 
I don't know if a frame buffer is the right way. From my point of view it is too time consuming to render a picture and then transfer it to the display. Next problem could be the required number of IO pins.

aenertia, 2018-09-04:
Well the easiest path is simply move that to something which does have a reasonable framebuffer/X11/wayland/whathave you. The dujour is octopi; but you can use a cheap tablet with a browser mounted with a USB port/old laptop etc etc. 

These screens are basically doing the above - and making that setup opaque and proprietary;  nextion has more resources than the MCU running the printer - none of which however are open designs. 

Basically my mantra is if I can't reasonably work with a completely open tool chain - the product is a trash fire.

alexxy, 2018-09-04:
There is opensource displays like PanelDue https://github.com/dc42/PanelDueFirmware with open schematics and firmware

aenertia, 2018-09-04:
> There is opensource displays like PanelDue https://github.com/dc42/
> PanelDueFirmware with open schematics and firmware

Sure - my point is ; it's probably best to define on the marlin side a
standard DTD(or similar thing) for UI Elements which can then have a
transform specific to each 'intelligent' screens bizarre and convoluted
implementation/makers personal preferences,  without ever caring about
anything more than defining things in the DTD.

Otherwise each 'smart screen' will require from scratch implementations

BigDi, 2018-09-04:
I got news of a new miniPC. They also sell a combination of screen and PC. This could be option for an open source implementation of a display. 
http://vocore.io/screen.html

thinkyhead, 2018-09-05:
For the curious, take a look at MarlinKimbra and/or MK4Duo, which supports the Nexion display.

BigDi, 2018-09-05:
I already took a look at MK and MK4Duo. Both projects uses Marlin 1.1.x Arduino libs which can't be used without any adaption with PlatformIO on my LPC1768 board.

AnHardt, 2018-09-05:
@BigDi 
https://www.youtube.com/watch?v=2-eclUz-RYI  ;-)

boelle, 2019-02-19:
@apballard could you submit a PR, of course its needs to pass inspection but it might good enough tht it can be added

guestisp, 2019-06-19:
So, Nextion support (or any other LCD) was abandoned ?

boelle, 2019-06-22:
@apballard ??

fiveangle, 2019-06-27:
His repo is therefor anyone to incorporate (also posted this at top of this thread): https://github.com/apballard/Marlin

weeelfi, 2019-09-29:
Apballard made a great implementation. I tested in a MegaRamps, and works very well. But Apballard didn't make any update.

What is needed to make for support Nextion display in marlin by default?

studiodyne, 2020-08-14:
**Idea**: Make a donation group , and ask @thinkyhead to make the module + a github for the marlin OEM firmware
Not infinite time wasted / not expensive if we are a big group,

Marlin really need , to have a DISPLAY , not 1000 differents displays with differents screen not complete ! , impossible to use a without the old lcd, because there is always something that missing , and that need to learn LINUX/PYTHON/JAVA and more to just add a screen with ' linear advance '!

Nextion is the solution

EvilGremlin, 2020-08-14:
You surely meant TJC displays?

thisiskeithb, 2020-08-15:
> You surely meant TJC displays?

TJC might make the panels, but they are commonly referred to/sold as "Nextion".

EvilGremlin, 2020-08-15:
I'm well aware :) He's fangirling over nextion too much and have no clue about real state of things.

Skorpi08, 2020-08-15:
https://www.youtube.com/watch?v=u2Hcwf2tZ7w

EvilGremlin, 2020-08-15:
Do you have it published somewhere?

Skorpi08, 2020-08-15:
Link in Description

Drakelive, 2020-08-26:
> 
> 
> Link in Description

Wow !!!!   
So, can we now integrate support into Marlin ?

Skorpi08, 2020-08-26:
there is nothing to integrate.
changes in M117.cpp is not must have.

futuretech208, 2020-08-28:
@Skorpi08 Nice work. What are the changes required to be made? I can see in your configuration file, there is no mention of Nextion display as a controller. Also, in actual Marlin folder, there is no header file included from Nextion. Is it not required?
I hope you built using PIO and not Arduino framework. It is getting compiled successfully in PlatformIO with your current settings.

Skorpi08, 2020-08-28:
https://github.com/Skorpi08/Nextion4Marlin/issues/8

futuretech208, 2020-08-28:
@Skorpi08 Thanks. Actually I went through your HMI file and understood by some reverse engineering. 😄
Basically Nextion display just acts as a serial command sender for Arduino. Technically anyone can modify the UI as per the firmware need after enabling certain functions. and each button after pressing should send the Gcode of the command.
Am I right in this?

Skorpi08, 2020-08-28:
yes

futuretech208, 2020-08-28:
@Skorpi08 Actually thanks a lot for giving that HMI file too. it was really of so much help. I am making my own screen UI now for 32 bit DUE. If anyone wants to make, there are few changes needed to be done. 
I followed this resource for the same.

https://nextion.ca/portfolio-items/nextion-iteadlib-and-due-step-by-step/ 

Thank you.

Skorpi08, 2020-08-28:
if possible, don't use Nextion Library.

futuretech208, 2020-08-28:
Ok, so you mean to say use "prints" function in Nextion Editor and write the code for each button. and then upload the code/.tft file to the touchscreen. Also, for SD print, where is SD card connected? Is it onboard SD card on the TFT or any other SD card? like on printer controller board?

Skorpi08, 2020-08-28:
I mean anything like this:
https://github.com/itead/ITEADLIB_Arduino_Nextion

Only onboard SD (on printer) works.

thisiskeithb, 2021-07-05:
Added in https://github.com/MarlinFirmware/Marlin/pull/21324

closed 2021-07-05 by thisiskeithb