MarlinFirmware/Marlin #7757: Compile error under Anet a8

kruuth, 2017-09-26:
### Compile Error
When I compile with `Anet A8` I get an error:
```
endstop_interrupts.h:124: error: missing binary operator before token "("

```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Phr3d13, 2017-09-26:
What branch, and can you upload Configuration.h and Configuration_adv.h

kruuth, 2017-09-26:
I'm using the latest version, downloaded from the main page.I'm attaching the two files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1334841/Marlin.zip)

Phr3d13, 2017-09-26:
There's 16 different branches, "latest version" doesn't give me enough information. Did you download a release (1.1.4, 1.1.5)? Did you download one of the branches as a zip?

kruuth, 2017-09-26:
My mistake.  I downloaded a zip of 1.1.5

fiveangle, 2017-09-26:
Your configs compile fine on bugfix1.1.x for me:

```
"C:\Users\djohnson\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -R .eeprom  "C:\Users\djohnson\AppData\Local\Temp\arduino_build_917996/Marlin.ino.elf" "C:\Users\djohnson\AppData\Local\Temp\arduino_build_917996/Marlin.ino.hex"
Multiple libraries were found for "LiquidCrystal.h"
 Used: C:\Users\djohnson\Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal
Using library LiquidCrystal at version 1.0.7 in folder: 
```C:\Users\djohnson\Documents\Arduino\libraries\LiquidCrystal 
Sketch uses 101072 bytes (79%) of program storage space. Maximum is 126976 bytes.
Global variables use 4160 bytes of dynamic memory.

kruuth, 2017-09-26:
OK I will DL it again and try it again

kruuth, 2017-09-27:
OK, just downloaded release 1.1.5 again.  I copied Configuration.h and Configuration_adv.h out of the Anet folder for A8.  I added the Anet A8 board to the hardware config and got the bootloader.  When those are selected, I get a new error:

`In file included from sketch\Marlin.ino.cpp:1:0:

D:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:257:26: fatal error: pins_arduino.h: No such file or directory

 #include "pins_arduino.h"

                          ^

compilation terminated.

exit status 1
Error compiling for board Anet V1.0.
`

Did I do something wrong here?

fiveangle, 2017-09-27:
bugfix1.1.x is where all the bugs found in 1.1.5 are fixed. bugfix* branches are always the ones that should be checked first if you ever have a problem.

kruuth, 2017-09-27:
Just tried bugfix1.1.x.  I'm still getting the same problem;

> Arduino: 1.8.4 (Windows 7), Board: "Anet V1.0"

In file included from sketch\Marlin.ino.cpp:1:0:

D:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:257:26: fatal error: pins_arduino.h: No such file or directory

 #include "pins_arduino.h"

                          ^

compilation terminated.

exit status 1
Error compiling for board Anet V1.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kruuth, 2017-09-27:
Ok built and installed.  Apparently there was an issue with the arduino install and not the firmware.  Sorry for the issue.

closed 2017-09-27 by kruuth

fiveangle, 2017-09-27:
I'm sorry you're struggling but as I mentioned, it compiled for me without issue with the Configs you provided.  Maybe an issue with your Arduino installation ?

kruuth, 2017-10-01:
Yes you're correct.  After a new install it worked fine.

fiveangle, 2017-10-01:
looks like our responses overlapped.  glad you got it sorted out !

Atatoth, 2017-11-23:
for me not :( i have this same problem

AnHardt, 2017-11-23:
Lucky you.
When you have the same problem - try the same fix.

So, exactly what problem?
What branch? From when?
Configuration?
What Arduino version?
...

Atatoth, 2017-11-23:
`endstop_interrupts.h:124: error: missing binary operator before token "("`

xyz axis with _MIN
branch last in bugfix release and 1.1.5, 1.1.6 release

arduino reinstalled not help

thinkyhead, 2017-11-24:
@Atatoth Please post your compile error in a new Issue. Thanks!