MarlinFirmware/Marlin #7758: constantly restart the Marlin

mkeyno, 2017-09-26:
constantly restart the Marlin 
following is the output of  Repetier-Host V 2.0.1
```cpp 
00:32:57.360 : ok T:start
00:32:57.361 : echo:Marlin 1.1.0
00:32:57.361 : Resend: N13 M111 S6*85
00:32:57.364 : echo: Last Updated: 2017-05-04 12:00 | Author: (KGC, default config)
00:32:57.368 : Compiled: Sep 27 2017
00:32:57.373 : echo: Free Memory: 3454  PlannerBufferBytes: 1232
00:32:57.377 : echo:EEPROM version mismatch (EEPROM=V13 Marlin=V37)
00:32:57.381 : echo:Hardcoded Default Settings Loaded
00:32:57.381 : echo:  G21 ; Units in mm
00:32:57.389 : echo:Filament settings: Disabled
00:32:57.390 : echo:  M200 D3.00
00:32:57.390 : echo:  M200 D0
00:32:57.390 : echo:Steps per unit:
00:32:57.393 : echo:  M92 X160.00 Y160.00 Z8000.00 E930.00
00:32:57.397 : echo:Maximum feedrates (units/s):
00:32:57.401 : echo:  M203 X200.00 Y200.00 Z3.00 E80.00
00:32:57.405 : echo:Maximum Acceleration (units/s2):
00:32:57.406 : echo:  M201 X9000 Y9000 Z70 E10000
00:32:57.414 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
00:32:57.418 : echo:  M204 P3000.00 R3000.00 T3000.00
00:32:57.426 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
00:32:57.430 : echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
00:32:57.434 : echo:Home offset:
00:32:57.434 : echo:  M206 X0.00 Y0.00 Z0.00
00:32:57.438 : echo:Material heatup parameters:
00:32:57.442 : echo:  M145 S0 H180 B70 F0
00:32:57.442 : M145 S1 H240 B110 F0
00:32:57.442 : echo:PID settings:
00:32:57.446 : echo:  M301 P22.20 I1.08 D114.00
00:33:00.469 : Error:Line Number is not Last Line Number+1, Last Line: 0
00:33:00.469 : Resend: 1
00:33:00.480 : ok
00:33:00.480 : Resend: N1 M110*34
00:33:00.480 : Resend: N2 M115*36
00:33:00.480 : Resend: N3 M105*36
00:33:00.480 : Resend: N4 M114*35
00:33:00.480 : Resend: N5 M111 S6*98
00:33:00.480 : Resend: N6 T0*60
00:33:00.480 : Resend: N7 M20*22
00:33:00.480 : Resend: N8 M80*19
00:33:00.480 : Resend: N9 M105*46
00:33:00.480 : Resend: N10 M105*22
00:33:00.502 : echo:SD card ok
00:33:00.547 : echo:SD card ok
00:33:00.555 : ok
00:33:00.555 : Resend: N11 M220 S100*81
00:33:00.592 : FIRMWARE_NAME:Marlin 1.1.0 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:  3D Printer EXTRUDER_COUNT:1 UUID:00000000-0912-0716-3464-000000000000
00:33:00.592 : ok
00:33:05.954 : ok T:start
00:33:05.954 : Resend: N12 M221 S100*83
00:33:05.959 : echo:Marlin 1.1.0
00:33:05.963 : echo: Last Updated: 2017-05-04 12:00 | Author: (KGC, default config)
00:33:05.967 : Compiled: Sep 27 2017
00:33:05.971 : echo: Free Memory: 3454  PlannerBufferBytes: 1232
00:33:05.975 : echo:EEPROM version mismatch (EEPROM=V13 Marlin=V37)
00:33:05.979 : echo:Hardcoded Default Settings Loaded
00:33:05.979 : echo:  G21 ; Units in mm
00:33:05.983 : echo:Filament settings: Disabled
00:33:05.983 : echo:  M200 D3.00
00:33:05.987 : echo:  M200 D0
00:33:05.987 : echo:Steps per unit:
00:33:05.991 : echo:  M92 X160.00 Y160.00 Z8000.00 E930.00
00:33:05.995 : echo:Maximum feedrates (units/s):
00:33:06.000 : echo:  M203 X200.00 Y200.00 Z3.00 E80.00
00:33:06.000 : echo:Maximum Acceleration (units/s2):
00:33:06.004 : echo:  M201 X9000 Y9000 Z70 E10000
00:33:06.012 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
00:33:06.016 : echo:  M204 P3000.00 R3000.00 T3000.00
00:33:06.024 : echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
00:33:06.028 : echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
00:33:06.032 : echo:Home offset:
00:33:06.032 : echo:  M206 X0.00 Y0.00 Z0.00
00:33:06.037 : echo:Material heatup parameters:
00:33:06.037 : echo:  M145 S0 H180 B70 F0
00:33:06.041 : M145 S1 H240 B110 F0
00:33:06.041 : echo:PID settings:
00:33:06.045 : echo:  M301 P22.20 I1.08 D114.00
00:33:09.068 : Error:Line Number is not Last Line Number+1, Last Line: 0
00:33:09.068 : Resend: 1
00:33:09.079 : ok
00:33:09.079 : Resend: N1 M110*34
00:33:09.079 : Resend: N2 M115*36
00:33:09.079 : Resend: N3 M105*36
00:33:09.079 : Resend: N4 M114*35
00:33:09.079 : Resend: N5 M111 S6*98
00:33:09.079 : Resend: N6 T0*60
00:33:09.079 : Resend: N7 M20*22
00:33:09.079 : Resend: N8 M80*19
00:33:09.079 : Resend: N9 M105*46
00:33:09.079 : Resend: N10 M105*22
00:33:09.079 : Resend: N11 M220 S100*81
00:33:09.101 : echo:SD card ok
00:33:09.145 : echo:SD card ok
00:33:09.153 : ok
00:33:09.191 : FIRMWARE_NAME:Marlin 1.1.0 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:  3D Printer EXTRUDER_COUNT:1 UUID:00000000-0912-0716-3464-000000000000
00:33:09.191 : ok
00:33:09.191 : Resend: N12 M221 S100*83
...

```

Tannoo, 2017-09-27:
Please follow the template that was deleted. That will help us help you.

mkeyno, 2017-09-27:
@Tannoo  actually I'm not sure it is bug or not and this just behavior that I couldn't find  the reason of restarting , I thought  past the output, could be helpful  to decode the problem

Phr3d13, 2017-09-27:
`echo:EEPROM version mismatch (EEPROM=V13 Marlin=V37)`
`echo:Hardcoded Default Settings Loaded`
I spotted these lines, it is the same for every restart. 
Also, Filling out the template would help us help you.

Tannoo, 2017-09-27:
M502
M500
M501

See if that helps.

Tannoo, 2017-09-27:
If you cannot send anything to it.. try just re-flashing, then send those.

thinkyhead, 2017-09-27:
Agreed. If the machine works fine without `EEPROM_SETTINGS` enabled, then `M500` followed by `M502` should clear up the issue.

mkeyno, 2017-10-01:
@Tannoo  I've been struggling  with this problem for couple of days  , notice if send any Gode  related to temperature it will chock & halt my Arduino board , even hard reset not working and LED1 (D13) start continuously flicking and it will stop when I cut the power. strange is my cnc work excellent with old version of Marline 1

Tannoo, 2017-10-01:
Try bugfix-1.1.x.

closed 2017-11-15 by thinkyhead