MarlinFirmware/Marlin #7759: [FR] Make UBL compile without a probe (for manual use)

deram, 2017-09-26:
### Feature Request
UBL has nice support for creating the mesh without probe (G29 I999\nG29 P2 B for example), but currently UBL can not be built without any probe.

Please make UBL applicable functionality available without probe setup.

Creating a NULL_PROBE that only returns Z=0 for all places on bed, could be usable, but disabling the probe only functionality would probably be safer.

Roxy-3D, 2017-09-26:
I agree....    This would be a good feature.

Tannoo, 2017-09-28:
I thought G29 P4 C1 could do the entire mesh without a probe.
Although,  I don't think it will compile without at least FIXED_PROBE defined.

Roxy-3D, 2017-09-28:
> I thought G29 P4 C1 could do the entire mesh without a probe.

Yeah.  That shouldn't need a probe.  That just needs the LCD Panel.  It grabs the encoder wheel for the editing.    But G29 P1 needs a probe currently.

deram, 2017-09-28:
At least for me, the 1.1.5 release won't compile UBL without one of the probes enabled.

I have made a very crude modification to make it build without probe. I'll tidy it up a bit in evening and publish it.

Roxy-3D, 2017-09-28:
> I have made a very crude modification to make it build without probe. I'll tidy it up a bit in evening and publish it.

Hopefully it  keys off of the current syntax to control code generation:
```cpp
#if HAS_BED_PROBE
#endif
```

deram, 2017-09-28:
#7787 

The modification needed to G26 is bit mystery to me. Can't see how it is working for anyone, other than AVR_AT90USB1286_FAMILY users.

Modification creates the null version of probe_pt, that is not compiled in Marlin_main.c when no probes are enabled.

Also the sanity check requiring probe for UBL should be lifted.

deram, 2017-09-28:
The config I'm using can be found in my config branch if needed to be referenced: https://github.com/deram/Marlin/commits/my_geeetech_config

Roxy-3D, 2017-09-28:
> Modification creates the null version of probe_pt, that is not compiled in Marlin_main.c when no probes are enabled.

I don't understand...   Where is the 'null version of probe_pt()' ?    Shouldn't we just not compile any code that needs the probe if HAS_BED_PROBE is undefined?

> Also the sanity check requiring probe for UBL should be lifted.

Agreed.

deram, 2017-09-28:
its in the chunk: 
```
+#if HAS_BED_PROBE
   extern float probe_pt(const float &lx, const float &ly, const bool, const uint8_t, const bool=true);
+#else
+  inline float probe_pt(const float &lx, const float &ly, const bool, const uint8_t, const bool=true) {
+    return NAN;
+  }
+#endif
```

But yes. The alternative is to actually go through the full UBL code and disabling everything requiring it.
This was my first try to figure out if this small change would result in the working parts getting usable.

Roxy-3D, 2017-09-28:
What file is that in?  I don't see that when I do a search for probe_pt()

deram, 2017-09-28:
Marlin/ubl_G29.cpp

Roxy-3D, 2017-09-28:
OK.    That is fine to get it going and to scope out the work.   But what we really want to do is put the 
```cpp
#if HAS_BED_PROBE
...
#else
  SERIAL_ECHO("UBL function not available due to missing Z-Probe.\n");
#endif
```

deram, 2017-09-28:
I agree. 
I've been reading also to the code doing the probings, and I think I can do another try with the method you are suggesting. There turns out to be just few of those places after all. 

I'll try to do another branch with this way implemented.

deram, 2017-09-28:
Any idea about the G26 part?
I'm getting following error when not addressing it at all.
```
sketch/G26_Mesh_Validation_Tool.cpp.o: In function `unified_bed_leveling::G26()':
sketch/G26_Mesh_Validation_Tool.cpp:238: undefined reference to `sync_plan_position_e()'
collect2: error: ld returned 1 exit status
```

It looks like sync_plan_position_e could have been normal function, but now in Marlin_main.c it is inline, and thus not directly able to be used in other files, even when linked together.

Roxy-3D, 2017-09-28:
Some times...  Even when you fix things, the error persists because Arduino doesn't recompile the file.   It might be good to kill Arduino and re-start it so everything gets compiled again.     But some how the linker isn't finding the function sync_plan_position_e().   So...   If that is an inline function, G26 needs to pull in that .h file.

deram, 2017-09-28:
That is inline function, defined in Marlin_main.cpp

deram, 2017-09-28:
I  think someone else has also hit this problem earlier, as there is line:
`  #if AVR_AT90USB1286_FAMILY  // Teensyduino & Printrboard IDE extensions have compile errors without this
`
I can't see this compiling anywhere else than on AVR_AT90USB1286_FAMILY MCU

Roxy-3D, 2017-09-28:
If the inline definition is in a .h file that is being pulled in...  Probably this declaration isn't needed:
```cpp
 void gcode_G28();
  void sync_plan_position_e();
  void un_retract_filament(float where[XYZE]);
```
Removing that might get rid of the problem.

deram, 2017-09-28:
the inline definition is here:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L774

deram, 2017-09-28:
Place where it is used (and link error is gotten) is here:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/G26_Mesh_Validation_Tool.cpp#L143

deram, 2017-09-28:
Everything seems to work (for me) when forcing to use the AVR_AT90USB1286_FAMILY code alternative, but what would be the correct way to handle this.

deram, 2017-09-28:
The if AVR... modification in G26 file is pretty new:
`9dda0227 (Bob-the-Kuhn   2017-06-15 00:32:05 -0500 139)   #if AVR_AT90USB1286_FAMILY  // Teensyduino & Printrboard IDE extensions have compile errors without this`

And the inline definition in Marlin_main.cpp about a year older:
`db5df9500 (Scott Lahteine           2016-07-24 13:27:09 -0700   774) inline void sync_plan_position_e() { planner.set_e_position_mm(current_position[E_AXIS]); }`

deram, 2017-09-28:
take 2 for UBL build without probe (G26 not yet included)
https://github.com/deram/Marlin/commit/64488f67c314cef0ddce954e3cd26f59cf05e872

deram, 2017-09-28:
Here's a take on the G26 part: https://github.com/deram/Marlin/commit/37ccc4abec5bc52ffe8def2f0574c728385e6313

deram, 2017-09-28:
Fixed some indentation and disabled building of unneeded functions
https://github.com/deram/Marlin/commit/1e46c7119bc8f2084afa4ca2a53986ed8dcc80b7

(compared to the first try with no code disabled at all, these modifications free 9260 bytes program memory when UBL built with no probes)

Roxy-3D, 2017-09-28:
> Everything seems to work (for me) when forcing to use the AVR_AT90USB1286_FAMILY code alternative, but what would be the correct way to handle this.

What version of Arduino are you using to compile this?   I'm wondering if can pull these lines out of G26.cpp now?
```cpp
  #if AVR_AT90USB1286_FAMILY  // Teensyduino & Printrboard IDE extensions have compile errors without this
    inline void sync_plan_position_e() { planner.set_e_position_mm(current_position[E_AXIS]); }
    inline void set_current_to_destination() { COPY(current_position, destination); }
  #else
    void sync_plan_position_e();
    void set_current_to_destination();
  #endif
```

deram, 2017-09-28:
I'm using Arduino 1.6.9 on OSX.

deram, 2017-09-28:
And it seems to build with all those lines removed.

Roxy-3D, 2017-09-28:
If I remember right...  Travis is a few generations behind the Arduino tool chain.   We need the code to compile in Travis also.   But it might make sense to comment out all but of that #if #else #end stuff and continue.   If Travis chokes on it, we may need to rethink how we deal with it.   

But having just the inline definitions available would be good.

deram, 2017-09-29:
Ok, I'll try to get these done again in the evening. I'll create a new pull request and add this UBL without Probe use-case in Travis configuration too, to see what it thinks about the changes.

fiveangle, 2017-09-29:
Why is that code an issue when used on non-AT90USB AVR ?

deram, 2017-09-29:
For some reason that won't compile on my Arduino 1.6.9.
I've got an Mega2560 based Geeetech board, so the code being compiled is the #else block.

Removing the whole #else block makes it compile just fine.

deram, 2017-09-29:
pushed current try to the PR. Travis seems happy.

deram, 2017-09-29:
There's still countless of lines unnecessary in LCD menu part of the code. Would it be feasible to design completely separate menu structure for the UBL without Probe scenario, as the majority of that menu is probe related?

fiveangle, 2017-09-30:
I recall now the AT90USB change that Bob put in, it was for AT90USB because at the time, apparently I was the only one having the issue on the Printrboard:  https://github.com/MarlinFirmware/Marlin/issues/7015

With the layout refactor it appears everyone else fell into the same boat.

Roxy-3D, 2017-09-30:
> There's still countless of lines unnecessary in LCD menu part of the code. Would it be feasible to design completely separate menu structure for the UBL without Probe scenario, as the majority of that menu is probe related?

There is a separate effort to make the LCD Panel support for UBL option-able.   I think the first pass of it will just turn it all off.   But we should be able to make a second pass through those changes and just leave most of it turned on but without a Z-Probe turn off the correct pieces of it.    Most of that code just encodes various G29 commands to be executed.   So if the probe isn't available, presumably the G29 code won't exist to do the various probed commands.

Roxy-3D, 2017-09-30:
Can you remove the lines?
```cpp
  #if AVR_AT90USB1286_FAMILY 
  #endif
```
It maybe that with everything declared inline it works now on all the compilers.

deram, 2017-09-30:
I already did that in the pull request https://github.com/MarlinFirmware/Marlin/pull/7787/files#diff-5dac0059c88a4755400e33c6c44db050 (which I updated to run travis check)

Roxy-3D, 2017-09-30:
Oh!   Ok!   I just checked...  Travis seems OK with it.    So my vote would be we don't add the extra kludge lines unless we start getting reports of  the problem coming back.

fiveangle, 2017-09-30:
since this is really about saving PROGMEM, probably best to save as many bytes as possible with shorter “Action not available without Z-Probe”.

Roxy-3D, 2017-09-30:
No...   This particular 'feature' is not about saving program memory.  This is about letting people that don't have a Z-Probe on the printer still setup and use UBL.    But because the whole Z-Probe subsystem (in Marlin_main.cpp) will disappear...  It could be implemented to save memory in the really tight 128 KB processors if the person still wanted to use UBL.

But... your wording does save memory and still communicates the message.   I'll change the text to that.   And we can discuss whether it makes sense to make it a function that gets called twice.

fiveangle, 2017-09-30:
since this [reduction of the string from the orignal suggestion] is really about saving PROGMEM, probably best to save as many bytes as possible with shorter “Action not available without Z-Probe”.

Roxy-3D, 2017-09-30:
It is changed to that in both places.
```cpp
+      #else
 +        SERIAL_ECHO("Action not available without Z-Probe\n");
 +      #endif // HAS_BED_PROBE
```

closed 2017-11-15 by thinkyhead