MarlinFirmware/Marlin #7776: [FR] [2.0.x] Ultimachine Archim (SAM3X8E) support

thinkyhead, 2017-09-27:
I've received a request to include formal support for this board.

This is a 32-bit RAMBo board. https://ultimachine.com/products/archim-1-0a

Pins definitions can be found at https://github.com/ultimachine/Marlin4Due/tree/Archim/Marlin

teemuatlut, 2017-09-28:
Is this a formal request from UltiMachine with possible sponsoring of a test board or is the developer required to purchase one on his own to get started?

RalphSch, 2017-09-28:
Just an other type of Due hardware, isn't it? If so it should work right out of the box with the given pin definitions .... if the Due code gets usable. Right at the moment there are still some issues preventing it from being used on a productive printer.

bobc, 2017-09-28:
Archim uses a customized Due core (see
https://github.com/ultimachine/Archim/wiki) which needs to be installed as
a new board in Arduino IDE. I looked, but I do not see PlatformIO support
for Archim, so it would need to be added to PlatformIO to build with that.

tonokip, 2017-09-30:
UltiMachine would like to provide+sponsor test boards to Marlin developers.  We can also provide samples of a new variant with tmc2130 drivers.  If you are interested then send an email request with a link to this discussion to info@ultimachine.com

teemuatlut, 2017-09-30:
You've said the magic word "Trinamic" right there. Email sent. Hopefully I'll hear something back from you guys.

jhitesma, 2017-11-02:
I recently came into possession of an archim board (Won a contest @Allted ran on the MPCNC forums) so with one on hand to play with I started looking into this.  I'm going off of @teemuatlut's  #8190 and the pin defs on the ultimachine wiki along with the marlin4due port.   

I haven't got it to compile yet so I haven't done a PR, but my branch is here: https://github.com/jhitesma/Marlin-Folger/tree/jth-archim  If anyone wanted to take a peek and warn me of any major mistakes I'm making or give me any tips.

teemuatlut, 2017-11-02:
What would your PR be about exactly?
The PR I have standing should provide a working setup along with pin definitions and anything else that's required. It was tested and used for printing at UltiMachine themselves.
There were some problems with recent changes to the Due HAL that broke compiling but at least some of those were already fixed.

jhitesma, 2017-11-02:
For the archim1.0a - basically just another pins file (and line in pins.h to reference it) and example config files since the pins are different than the archim2.0.

So the big thing is just the different pins defs, and the different config files since it doesn't have those sexy trinamic's ;)  

The other other changes are already in your PR - so I'd wait for that to merge first.  It wouldn't be that hard to add to your PR but that would make your PR less atomic so I was hesitant to suggest it.

jhitesma, 2017-11-03:
I see that #8190 merged...but when I tried to build it it complains about the tmc2130 driver files not being present.

I rebased my work on archim1.0 this morning and found a few other things that have me scratching my head.

1) the define for SDSS in pins_ARCHIM2.h is inside the LCD ifdef's...but looking at other marlin2.0 pins files it appears this is now required even when LCD's aren't enabled.  In my code I got errors with it in the LCD section even if I configured an LCD.  Moving it to a miscellaneous section above the LCD section like I saw done in other pins files solved that.

2) There are no current control pins defined on archim2.  I'm guessing this is because it's going to be handled through SPI on the TMC2130's but don't see any code to address that.  So it appears that as currently merged there's no way to control the current on archim2.

On archim1.0 I looked at the marlin4due port and brought over the pin definitions for current control - but they broke compilation.  Looking at the code it appears current Marlin2.0 only supports 3 channels of current control, a combined XY channel, Z, and a single extruder.  Marlin4Due was using 5 channels (X,Y,X,E0,E1).   I'm looking into modifying 2.0 to support this - but the way current control is handled has changed substantially and I'm still trying to figure out the best way to approach that.   I was able to build for archim1.0a with current control disabled - but I'm not about to try it on a board without that!  

BTW - if there's a better place to discuss this please let me know.  (I can be reached through my same username on gmail if someone wanted to discuss anything in email) I could open a PR with my work so far but I assume it's prefered that it be more or less complete before I do that so I haven't yet.

teemuatlut, 2017-11-03:
I think here is fine.

The TMC2130 current control is handled by the library and Marlin makes use of that. The digipot or pwm current controls are seperate from anything Trinamic and you may need to expand their functionality.

fiveangle, 2017-12-15:
> Archim uses a customized Due core (see
> https://github.com/ultimachine/Archim/wiki) which needs to be installed as
> a new board in Arduino IDE. I looked, but I do not see PlatformIO support
> for Archim, so it would need to be added to PlatformIO to build with that.

Is anyone working on this aspect or is there any work-around to support compiling in PlatformIO ?  Without that part, the board cannot be included for regression testing since TravisCI for 2.0.x branch currently uses PlatformIO (see #8798)

jhitesma, 2017-12-15:
I started to look into that last month but have been crazy busy with life stuff and didn't have any time to even try.  But there is info on adding a custom board here:
http://docs.platformio.org/en/stable/platforms/creating_board.html

teemuatlut, 2017-12-15:
@tonokip Care to weigh in on this? It's possible that just for compile test purposes all it needs is the compile time definition for `ARDUINO_SAM_ARCHIM`.

marcio-ao, 2018-03-13:
Just to chime in on this thread, we are working on a printer release using the Archim 2. So we would be willing to provide assistance. This thread hasn't been updated since December, so curious as to what the status currently is.

teemuatlut, 2018-03-13:
I'm using the board on a CoreXY build.
The drivers seem to work without a problem and all the TMC related commands are alright. The DUE core seems to work fine as well.
The problems I have with the printer are not related to the motherboard.

There hasn't been updates because everything should be alright.

marcio-ao, 2018-03-13:
That's good to hear. Are there any timelines for an initial Marlin 2.0 release? What are the major issues holding it back? Is it still bleeding edge, or just slightly-give-you-a-bruise edge?

teemuatlut, 2018-03-13:
@thinkyhead can probably give you a better answer.

But what I know is that v1.1.9 should be the next release and is intended to be the last one in v1.1.x series. Supposedly there are some issues that Scott would like to fix first so it's possible that the plan will change. The bugfix-2.0.x branch should be on par with 1.1.x as far as AVR is concerned and I believe that is the primary goal. From there we can again focus on one branch only and better test the 32bit platforms. My hope actually is that the developers would already be running the v2 branch on their machines and I have not seen issues on my main machine which is running a standard ATMega.
I do not expect the 32bit HALs to have feature parity with AVR when v2 is released but if the main features work, and I believe they do, that should be a very good starting point for wider adoption.

thinkyhead, 2018-04-24:
The initial 2.0 release will be done when there's a critical mass of boards and features at least at beta status. Some features won't be available for some platforms until we have 32-bit support for their top-level APIs (e.g., Neopixel). Keep an eye on #7076 for which boards are beta and which are far from completion.

marcio-ao, 2018-04-24:
@thinkyhead : Should the Archim 2.0 be added to the list? This is the board I am working on.

thinkyhead, 2018-05-03:
If I didn't mention it elsewhere, the board is now on the list.

jhitesma, 2018-05-03:
Might want to update the list to clarify Archim vs. Archim 2.0 support.  I know there's been a lot of work on archim 2.0 but  I haven't seen anything on the original Archim yet.

1.0 is almost the same but instead of tmc2130 drivers it has drv8825's and last time I tried to get mine going ran into an issue with setting current (though it looks like Marlin2.0 may have changed that enough since I last checked that it will no longer be a roadblock...previously there was only a way to set a single x/y current not separate x/y as archim needed) 

Hopefully next week I'll have some time to see if I can make progress on my 1.0 board again.

thinkyhead, 2018-05-04:
@jhitesma — Well, do keep us posted on things that need to be fixed. We'd like to get as many common boards supported as possible.

Allted, 2018-11-12:
I think this is ready to go #12404 

I tried and tested everything I could think of. Usb works now, LCD works, step rates and driver current all check out.

@tonokip, thanks for doing all the heavy lifting!

aznoohwee, 2019-01-15:
I'm doing a little investigative work trying to get this board up and running on @Allted 's MPCNC branch.  The build from the Arduino GUI is giving a different output then the platformio build [see my branch](https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...aznoohwee:feature/MPCNC_Archim1_PlatformIO_Target?expand=1#diff-237285c047dad808f73698c139144e1a).

My initial review of the code seems to indicate that the fastIO pin defs for this board differ from the UltiMachine add-on configuration available in the UI.  It seems that the Arduino GUI uses the board configuration from UltiMachine and produces a working build whereas the fastIO defs used by platformIO generate an incorrect output.

Bugasu, 2020-03-04:
So attempting to build this in a fresh VSCode copy of this branch, I used ENV:DUE as labelled. However, it would appear there is no definition of ARDUINO_SAM_ARCHIM when building via platformIO and therefore some pins are left undefined for fastio.h and the build fails.

Is this intended behavior? Am I supposed to modify the environment to include this define? I notice a few other options that switch based on hardware variant archim in the makefile, and I'm wondering if those also need to be created.

colinoflynn, 2020-09-04:
The platformIO pin problem is I think also the cause of bug #19215 and #18816, although the build now succeeds (but has silent errors, the best kind of errors). The `variant.cpp` file is needed at least to get pins working, but I assume there is more here...

sjasonsmith, 2020-11-10:
Based on recent changes from a couple users I believe Archim boards are currently working. I'm closing this feature request. Please reply with details if you feel this was incorrect.

closed 2020-11-10 by sjasonsmith