MarlinFirmware/Marlin #7782: BLtouch & Standard HS-645 Servo Running Concurrently - Azteeg X3 Pro

Printalect, 2017-09-28:
------------------------------------------------------------
### Bug Report - (TBD)

Intro: I am not sure if this is the correct place to post this since I can't say if this is a legitimate issue, or if I'm just simply missing something. Board =  Azteeg X3 Pro.

Issue: I am trying to get a BLTouch and a servo for the "Switching Nozzle" function working concurrently. They work separately when either or is connected, just not at the same time.

Assist Needed: I can likely figure this out, but I am confused about one thing... If both of these are essentially servos, and all we are doing is defining servos to equal 1 or 2, then how is Marlin deciding which pin to use for either the BLTouch or the actual Servo on for the switching function? How is it differentiating the two?

... (from: Configuration.h) 
#define NUM_SERVOS 2

.... (from: pins_AZTEEG_X3_PRO.h)
#undef SERVO0_PIN 
#undef SERVO1_PIN 

#define SERVO0_PIN  31  // Intended pin for the Servo
#define SERVO1_PIN  32  // Intender pin for the BLtouch

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1339281/Configuration.zip)

Roxy-3D, 2017-09-28:
Even on a RAMPS board a servo can be too much power for the board to supply.  And you are asking your board to supply two power pigs?

Printalect, 2017-09-28:
The Azteeg is stated to supply 5v at 1amp. I've tested both 5v pins on the board and they both get (5.03V).

The BLTouch is stated to use a max current of 300mA and the HS-645, which is under minimal load at the moment, pulls approx 400mA under no load. I would think this is fine, yet I don't have a great degree of knowledge on the topic. I'm not sure what you mean by glitching the board logic. Is there a way to test for this?

I am currently testing through the pins on the AX3P, since this was an issue for #6756. So far, I have found that when setting two servo pins (pins_AZTEEG_X3_PRO.h ), as is shown above, the first pin is always recognized by the servo, but the second pin is never found by the BLTouch.

I have been testing between pins 31, 32, and 47. 

Board illustration for reference:
[image]

Roxy-3D, 2017-09-28:
OK...  The reason for the question is on some boards (like the RAMPS) the servo takes so much power the microprocessor gets confused (from the low voltage) and resets.     If that is not happening, that is good.

The Z-Probe servo is controlled in your Configuration.h file and is typically Server 0.
```cpp
  #define SWITCHING_EXTRUDER_SERVO_NR 0
```

Grogyan, 2017-09-28:
These low voltage regulators typically have the wrong output current rating, I don't know why people keep putting a 1A LDO regulator and get weird things happening because the poor regulator can't supply the required current while maintaining 5V out.
Suggest putting another regulator for the servo line, 1.5A should be enough, if not, then a 2A.

Printalect, 2017-09-28:
Perfect, thanks. That's the missing detail. The Servo is switching on Tool Changes and the BLTouch responds to all M280 commands.

I had researched on how to point different servo options to certain pins, or at least map them in some way, whether doing switching nozzle, BLTouch, or the filament runout sensor (which I think can occupy a servo pin).... any of them simultaneously. And I am guessing one way to "map" these pins to at least one servo, the switching nozzle/extruder, is through ....Extruder_Servo_NR. That makes sense. 

So the BLTouch will always take precedent on the first Servo Pin?

Also, this is working on a test bench. I'll move this to an actual printer to confirm that everything is working as it should.

JerryDurand, 2017-09-28:
Don't forget to account for Safe Operating Area and general heat 
dissipation.  A standard TO-220 package (LM7805 type regulator) is good 
for about 1 Watt in free air, a little more with a fan.

If you're dropping from 12V, then your maximum current is:
Amps = (Watts / (Vin - Vout)) - Quiescent_Current
so, for the example LM7805

(1 Watt / (12 V - 5 V)) - 0.01 A = 0.132 Amp.

closed 2018-02-09 by thinkyhead