MarlinFirmware/Marlin #7784: Auto Level consistently gives probe failed

aaacomputersolutions, 2017-09-28:
### Bug Report
- Description: Probe works but autolevel fails with ERROR: PROBING FAILED
- Expected behaviour: --- Autolevel working?
hictop 3dp-08 
Marlin 1.1.5
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1341202/Configuration.zip)

`echo:M502
echo:Hardcoded Default Settings Loaded
ok
echo:M500
echo:Settings Stored (498 bytes; crc 14897)
ok
echo:M501
echo:V40 stored settings retrieved (498 bytes; crc 14897)
Unified Bed Leveling System v1.01 inactive.

Unified Bed Leveling initialized.

UBL System reset()
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z398.00 E94.50
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Unified Bed Leveling:
echo:  M420 S0 Z10.00

Unified Bed Leveling System v1.01 inactive.

Active Mesh Slot: -1
z_offset: 0.000000
EEPROM can hold 8 meshes.

echo:Material heatup parameters:
echo:  M145 S0 H210 B70 F0
  M145 S1 H230 B100 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:Z-Probe Offset (mm):
echo:  M851 Z0.00
ok
echo:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:105.00 Y:135.00 Z:0.00 E:0.00 Count X:8400 Y:10800 Z:0
ok
echo:G29 P1
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:Probing failed
echo:busy: processing
echo:busy: processing
Error:Probing failed`

aaacomputersolutions, 2017-09-28:
I've also tried 3 point and Probing Failed on that as well...probe looks like its working (light lights and firmware reports triggered when it does)

dejanr, 2018-01-16:
I am getting the same thing on bugfix 2.0.x branch

```
echo:G30 A
  current_position=(0.00, 0.00, 210.00) : bracket_probe_move
>>> probe_pt(40.00, -20.00, no stow)
  current_position=(0.00, 0.00, 210.00) :
>>> do_blocking_move_to(0.00, 0.00, 145.69)
  destination=(0.00, 0.00, 210.00) : set_destination_from_current
  destination=(0.00, 0.00, 145.69) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 145.69) : zone border move
  current_position=(0.00, 0.00, 145.69) : xy move
echo:busy: processing
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 145.69) : set_probe_deployed
deploy: 1
  current_position=(0.00, 0.00, 145.69) : >>> run_z_probe
  current_position=(0.00, 0.00, 145.69) : >>> do_probe_move
>>> do_blocking_move_to(0.00, 0.00, 100.00)
  destination=(0.00, 0.00, 145.69) : set_destination_from_current
  current_position=(0.00, 0.00, 145.69) : xy move
  destination=(0.00, 0.00, 100.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, 100.00) : z lower move
echo:busy: processing
<<< do_blocking_move_to
  current_position=(0.00, 0.00, 100.00) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 100.00) : <<< do_probe_move
  current_position=(0.00, 0.00, 100.00) : >>> do_probe_move
>>> do_blocking_move_to(0.00, 0.00, -10.00)
  destination=(0.00, 0.00, 100.00) : set_destination_from_current
  current_position=(0.00, 0.00, 100.00) : xy move
  destination=(0.00, 0.00, -10.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -10.00) : z lower move
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to
  current_position=(0.00, 0.00, -10.00) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, -10.00) : <<< do_probe_move
>>> do_blocking_move_to(0.00, 0.00, -5.00)
  destination=(0.00, 0.00, -10.00) : set_destination_from_current
  destination=(0.00, 0.00, -5.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -5.00) : z raise move
  current_position=(0.00, 0.00, 0.00) : xy move
  destination=(0.00, 0.00, -5.00) : prepare_uninterpolated_move_to_destination
  current_position=(0.00, 0.00, -5.00) : z lower move
echo:busy: processing
<<< do_blocking_move_to
<<< probe_pt
Error:Probing failed
[ERROR] Error:Probing failed

  current_position=(0.00, 0.00, -5.00) : bracket_probe_move
```

I have tried inverting zprobe inverting but same. I have set

wormiedk, 2018-01-19:
You need to disable 
//#define Z_MIN_PROBE_ENDSTOP
and enable
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

That worked for me :)

closed 2018-02-14 by thinkyhead