MarlinFirmware/Marlin #7785: Adjusting Feedrate does not Affect Actual Printing Speed

kevintong116, 2017-09-28:
I'm using the 1.1.x bugfix version kossel generic
When I turn the encoder on my 2004 LCD, the feedrate reading changes.
But the actural printing speed of the printer doesn't changed at all.
What's the problem?
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1341256/config.zip)

AnHardt, 2017-10-01:
More infos please.
During all automated moves (homing, probing, ...) the feedrate-factor is ignored.
If printing already fast - Marlin will not exceed :
```
/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 200, 200, 200, 200}
```

kevintong116, 2017-10-02:
I did not change the max feed rate.
I'm just printing at 45mm/s. I tried to set the factor from 10% to 200%, and the printing speed still didn't change.
And seems that the z babystepping also not talk effect.

kevintong116, 2017-10-04:
I have tried using the newest bugfix but still not working

kevintong116, 2017-10-16:
#7948 Solve the problem

closed 2017-10-16 by kevintong116