MarlinFirmware/Marlin #7788: Make printer execute script when is poweredon

batata004, 2017-09-28:
Hi,

I loved the new feature of Marlin that allows user to add custom scripts the LCD menu. That's nice! But everytime I boot up my printer I need to execute a script (it's a routine that heats up things, auto level, home...).

Is it possible to make the printer everytime it is powered on it executed a script? Is this feature already implemented? I belive so cause I think many people execute the same procedure when they usually turn on their printer, but I searched at google and couldnt find it.

teemuatlut, 2017-09-28:
[auto0.g file on an SD card](https://github.com/MarlinFirmware/Marlin/wiki/Autostart)

batata004, 2017-09-28:
@teemuatlut that's nice, really nice! But if I already have my printer turned on and insert the card it will not automatically play the auto0.g file, right?

fiveangle, 2017-09-29:
No

http://bfy.tw/ECOX

thinkyhead, 2018-02-09:
Careful. You wouldn't want Marlin to do a `G28` if it rebooted in the middle of a print.

closed 2018-02-09 by thinkyhead