MarlinFirmware/Marlin #7789: UBL does not compensate

floyd871, 2017-09-28:
### Bug Report
- Description: UBL does not compensate
- Steps to reproduce:
I haved followed the steps descriped here: http://marlinfw.org/docs/features/unified_bed_leveling.html
SENDING:G29 T
Bed Topography Report:
(0,4)                           (4,4)
(25,175)                        (175,175)
 -0.323   -0.211    0.027   -0.117   -0.402
 -0.249   -0.019   -0.089   -0.222   -0.348
 -0.208    0.014   -0.032  [-0.172]  -0.378
 -0.193   -0.093   -0.051   -0.192   -0.361
 -0.243   -0.097   -0.031   -0.207   -0.407
(25,25)                            (175,25)
(0,0)                             (4,0)
-G28, G1 Z10 X100 Y100 F3000, G1 Z0
put a paper below the nozzle and move the nozzle down (M211 S0) till the paper is touched
increase the x axis by 1 mm till it reaches 174, the paper should always be touched, i see the z axis moving but this is not enough. i have to increase z again by 0.2 
Then suddenly at x=175 z moves quite a bit . i have to compensate z again manually by 0.4.
same issue when i decrease x from the middle. at the outer bed i have to compensate z again by 0.2 that the nozzle touches the paper again.t

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1342030/Marlin.zip)

Roxy-3D, 2017-09-28:
If you move the nozzle to (X,Y) = (26,26) and do
- G29 A
- G1 Z 1.0
- G1 X170 Y170

Do you see the Z-Axis correcting as the nozzle moves across the bed?

floyd871, 2017-09-29:
Yes, the Z Axes is correcting across the Bed.
I am a bit curious, When i move the nozzle outside the probing area (above (25,25) (175,175)) the nozzle makes a jump in Z axis correction. But only when i move X 174...175... and Y 174...175
Not the other corner X 26...25...or Y 26..25

Roxy-3D, 2017-09-29:
When you leave the mesh...  No correction is applied.   So on the right side, for example, you go from a correction of -.35 or -.4 mm to 0.0 mm.   You are seeing a jump because UBL doesn't know what the correction should be and none is being applied.     

Your mesh should cover the area you are using.   You should probably re-setup your printer to make that happen.

floyd871, 2017-09-29:
Ahh, okay i did'n't know that. I thought there is a kind of extrapolation outside the mesh area. I will reconfigure that.
But inside the mesh there is still too little compensation.

Roxy-3D, 2017-09-29:
> But inside the mesh there is still too little compensation.

Have you printed a calibration cube in the center of the bed with no bed leveling active?  Hopefully you have.   The reason being, then we know all the axis are moving the correct amount when told to move.

But, once the new mesh is defined and filled...   We will use G26 to make sure everything is accurate.   If you post a picture of your G26 patterns, I'll coach you on what I would do given what I'm seeing.

closed 2017-10-04 by floyd871