MarlinFirmware/Marlin #7790: [FR] Remote Serial Display Functionality for External LCD Controllers

Deneteus, 2017-09-28:
I would like to see some sort of external serial display functionality so I can do stuff like add an Android display as an external monitor for multiple printers. The ability for Marlin to output data to an external display allows boards with lower available memory to still use remote displays which will lower the size of the code needed for basic monitoring functionality. Wanhao i3 Plus does something like this and so does Kimbra by using external LCD's with their own dedicated storage/controller. 

I am envisioning something like loading an android app on my LG or Samsung Smart TV and being able to monitor print status without needing OctoPI or an external controller since remote control is built into the TV itself.

pyr0ball, 2017-09-29:
I believe the DuetWifi also does this, but that's an ARM32 processor board from the getgo

Grogyan, 2017-10-02:
Yes, CAN bus would be ideal for these types of displays.
Hence #7735

AnHardt, 2020-07-08:
Original request is done in several ways.

closed 2020-07-08 by ellensp