MarlinFirmware/Marlin #7791: fan controller error

vMeph, 2017-09-29:
for some reason i always get a error when enable fan controller on Marlin-bugfix-2.0.x
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 44  // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif

e get the following  error
C:\Users\vMeph\AppData\Local\Temp\arduino_build_537699\sketch\src\feature\controllerfan.cpp: In function 'void controllerfan_update()':

C:\Users\vMeph\AppData\Local\Temp\arduino_build_537699\sketch\src\feature\controllerfan.cpp:33:9: error: 'X_ENABLE_READ' was not declared in this scope

     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0

         ^

C:\Users\vMeph\AppData\Local\Temp\arduino_build_537699\sketch\src\feature\controllerfan.cpp:33:41: error: 'Y_ENABLE_READ' was not declared in this scope

     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0

                                         ^

C:\Users\vMeph\AppData\Local\Temp\arduino_build_537699\sketch\src\feature\controllerfan.cpp:33:73: error: 'Z_ENABLE_READ' was not declared in this scope

     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0

                                                                         ^

C:\Users\vMeph\AppData\Local\Temp\arduino_build_537699\sketch\src\feature\controllerfan.cpp:33:105: error: 'thermalManager' was not declared in this scope

     if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0

                                                                                                         ^

C:\Users\vMeph\AppData\Local\Temp\arduino_build_537699\sketch\src\feature\controllerfan.cpp:34:12: error: 'E0_ENABLE_READ' was not declared in this scope

         || E0_ENABLE_READ == E_ENABLE_ON // If any of the drivers are enabled...

            ^

Using library U8glib at version 1.19.1 in folder: C:\Users\vMeph\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

vMeph, 2017-10-01:
Any ideas how to fix

bobc, 2017-10-02:
controllerfan.cpp is missing some includes, try adding lines marked "+" as follows: 
```
 #include "../inc/MarlinConfig.h"
 
 #if ENABLED(USE_CONTROLLER_FAN)
 
+#include "../module/stepper_indirection.h"
+#include "../module/temperature.h"

 void controllerfan_update() {
   static millis_t lastMotorOn = 0, // Last time a motor was turned on
                   nextMotorCheck = 0; // Last time the state was checked
   const millis_t ms = millis();
```

vMeph, 2017-10-02:
@bobc  that have fixed now is working
will that lines missing be apply on main code?
or is just a problem on my side?

thanks alot for your help rox!!!!

closed 2017-10-04 by vMeph

fiveangle, 2017-10-07:
#7877 submitted (probably best not to close an issue until there is a committed resolution ;)

vMeph, 2017-10-07:
ya thank you

insanity67, 2018-05-03:
I get this error too on 1.1.8 bugfix with controller enabled.
Looks like I dont have a controllerfan.cpp, so how can I fix this?

`Marlin_main.cpp:13567: error: 'class Temperature' has no member named 'soft_pwm_amount_bed'

       if (X_ENABLE_READ == X_ENABLE_ON || Y_ENABLE_READ == Y_ENABLE_ON || Z_ENABLE_READ == Z_ENABLE_ON || thermalManager.soft_pwm_amount_bed > 0`

thinkyhead, 2018-05-04:
I've patched the issue in the current `bugfix-1.1.x` so it should compile for you now.

insanity67, 2018-05-04:
thank you @thinkyhead, works like a charm 👍