MarlinFirmware/Marlin #7796: Re-Arm discussion..

gunnarfreyr, 2017-09-29:
So i noticed in the readme file for Marlin 2.0 that one of the current HAL's was for Re-arm .. Then afer googling the matter for a day or two , i saw some people say that there has been a successfull compile onto a re-arm board, and that some people have had issues with that.. 

However there does not seem to be much info on how to compile it or anything else for that matter.. 

A) Can anybody here confirm that anybody has a running Marlin 2.0 on a Re-Arm board ? 
B) Could anybody explain roughly the process of how they got it up and running ? 
C) Are there any plans for "full" Re-Arm support with Marlin 2.0 (for 45$ Re-Arm seems like a good option for getting into 32bit boards)

So far I've been trying to use PlatformIO to complie Marlin2.0, but cant import the M2 as a project for mbed lpc1768 , cause it does not support arduino framework, and since there is basicly no info on this out there, i figured i would post this here and get this sorted out :)

Roxy-3D, 2017-09-29:
We are using the Re-ARM board as the primary 32-bit development platform.  But the effort is already starting to fan out onto other 32-bit boards.  Especially the STM boards.  Right now, Re-ARM is the most supported 32-bit platform.

People are printing with Marlin on Re-ARM boards right now.   But not all features work.   Most of your questions about how to get the development platform up and running are in this thread: 
  https://github.com/MarlinFirmware/Marlin/issues/7076   Read any post that references PlatformIO or PlatformIO-IDE

p3p, 2017-09-29:
Hopefully there will be much more concise instruction by the time we get to a Marlin 2.0 release, we are currently in alpha .. maybe beta ;) testing so it's expected that it will not be perfect and most likely you will have to assist with testing and debugging rather than having a working printer, saying that the developers (there's a few of us at this point) working on Re-ARM (it's general LPC1768/9 or will be when finished) have got to the point where most Marlin features can be enabled and we are getting good prints.

The LPC1768 platform is not an Arduino platform so needs to be built with PlatformIO IDE, once that is installed correctly (some people struggle with the setup others it just installs smoothly) you just open the Marlin project and select the appropriate platform to build, then copy the built binary onto the Re-ARMS usb storage.

gunnarfreyr, 2017-09-29:
Thanks for the info guys :) I dont mind doing some "beta testing" , i think its more fun trying new things on my printer than printing.. And since i have a Re-arm board, and found out very quickly that Smootieware kinda sucks, i wanted to use the board for something, and then i found out about Marlin 2.0, and since i use and like Marlin, i figured why not go for that :) 

However when i try to import the Marlin 2.0 as a project for the mbed lpc1768, PlatformIO just refuses and says that it's not compatible with arduino framework.. But im gonna read the #7076 thread and see if that helps me figure that out :)

p3p, 2017-09-29:
It should be simple as selecting open project and pointing it at the platformio.ini file in the repository.

fiveangle, 2017-09-30:
You have to run the development version of PIO core AFAIK.  But it sounds like your issue is you may not have your board+PIO env matching correctly ?  You have to set the Re-ARM board (e.g. "RAMPS_14_RE_ARM_EFB")  in config, plus the Re-ARM PIO environment from platformio.ini

gunnarfreyr, 2017-09-30:
Well ..  since i just found out about PlatformIO yesterday, im pretty sure im doing something wrong :)
My steps were : 
Open Atom and install PlatformIO 
Selected "Import Arduino Project" and selected the Marlin-bugfix-2.0.x and "NXP mbed LPC1768" 
and then it says that it's not compatible with Arduino framwork 
Started to google what i did wrong :)

p3p, 2017-09-30:
Like I said above you need to open the PlatformIO project not import the Arduino project, then setup the configs (using the motherboard @fiveangle said) then in menu find "Run other target" and scroll to the bottom of the list to build Re-ARM.

Roxy-3D, 2017-09-30:
How to get PlatformIO-IDE to load a project is up in the first half of the #7076 issue.   And exactly how to get the first version of Marlin boot strapped so you can use the USB cable instead of SD-Cards.

556duckvader, 2018-01-01:
Are there anymore instructions for getting this installed on a Re-Arm.  I have tried to follow the instructions and I keep getting errors.  Also there is no Re-Arm listed in the platformio.ini file in the 2.x download.

p3p, 2018-01-01:
Choose LPC1768, it was changed to match the other platforms its the MCU type not the board name that is built for, in the config the board is RAMPS_14_RE_ARM_EFB (or the ramps version you use)

556duckvader, 2018-01-01:
This is what I am getting when I go to compile it.
Marlin\src\HAL/HAL_endstop_interrupts.h:51:4: error: #error Unsupported Platform!
#error Unsupported Platform!
^~~~~
Marlin\src\Marlin.cpp: In function 'void setup()':
Marlin\src\Marlin.cpp:833:30: error: 'setup_endstop_interrupts' was not declared in this scope
setup_endstop_interrupts();
^

*** [.pioenvs\LPC1768\src\src\Marlin.o] Error 1
 [ERROR] Took 5.86 seconds
 
 [SUMMARY]
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment anet10                      [SKIP]
Environment DUE                         [SKIP]
Environment LPC1768_debug_and_upload    [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment LPC1768                     [ERROR]
 [ERROR] Took 5.86 seconds

p3p, 2018-01-01:
> Marlin\src\HAL/HAL_endstop_interrupts.h:51:4: error: #error Unsupported Platform!
> #error Unsupported Platform!

That means exactly what it says, Endstop Interrupts are not currently supported on the LPC1768 platform and need disabled in your config, because of the speed of the MCU its not that bad a thing they will probably be implemented at some point.

556duckvader, 2018-01-01:
Ok got it to load and I am doing some testing to make sure everything works.  Is there a list of  what features are not available when using Re-Arm/RAMPS.

closed 2018-02-11 by thinkyhead

AJMOZ, 2018-03-31:
Please, anyone building for ARM - PLEASE build it without the Arduino framework. There are full ARM IDE freely available with in circuit debug etc. Arduino was nice for 8 bit, for 32 it really sucks.

Roxy-3D, 2018-04-01:
> Please, anyone building for ARM - PLEASE build it without the Arduino framework. There are full ARM IDE freely available with in circuit debug etc. Arduino was nice for 8 bit, for 32 it really sucks.

I wish Arduino could build the Re-ARM images...   But it can't.      Is there a workable IDE for ARM other than Platform-IO ?

AJMOZ, 2018-04-02:
For atmel:avrstudio, for stm32:atollic and I'm sure there are others using
gcc also

--= sent from mobile =--

thinkyhead, 2018-04-02:
We hope that experts and users of those tools will contribute as the 2.0 branch progresses.

Roxy-3D, 2018-04-02:
@AJMOZ 
> For atmel:avrstudio, for stm32:atollic and I'm sure there are others using `gcc` also

What needs to be done to get the bugfix_2.0.0 branch to compile in Atmel's AVR-Studio ?   Is that something you can take a quick look at and see if it can make an image?   A lot of people would like to ditch Platform-IO.   (Me included!)

ModMike, 2018-04-02:
Are there any step by step instructions for using Platform IO with Re-Arm?  I'm getting bits and pieces everywhere and some instructions refer to what seems to be deprecated or changed functions such as process (I only see finish).

Roxy-3D, 2018-04-02:
Its a long thread...  But #7076 has all the information to get Platform-IO building Re-ARM images.
There is not currently a concise, short, simple instruction list.

ModMike, 2018-04-02:
Roxy,

Thank you for the quick response!

Length is ok but it does not work.  Example:

Once the PlatformIO home/welcome page is displayed
Click on the +New Project button. Select Other under Chose the directory and then enter the directory ABOVE Marlin.
Click -- choose a board -- and scroll down to the Platform NXPLPC area and click on NXP mbed LPC1768.
Click the Process button. Hopefully no errors will pop up.
In the extreme lower left corner of the Atom window there should be something with PIO Build in it. Click on that.
In the window that pops up enter Re-ARM in the search window.
Click on PIO Build (Re-ARM). That'll open up a window where it will show the compiler output.
If you have some compiler errors the window will stay open. Make your corrections and then, to re-run the compiler, click the lightning bolt button on the upper right of the window.
Once it says SUCCESS the executable will be the file firmware.bin in the directory .pioenvs\Re-ARM. Copy that to an SD card and insert in into the SD card reader on the motherboard. After cycling power (reset won't do it) the new code will be loaded and run.

First off new project asks you to choose board.  There is no process button (is it "finish now")

ModMike, 2018-04-02:
By the way, can I use the Marlin 1.18 config.h and config_adv.h with Marlin 2.0 or do I need to start from scratch?

Roxy-3D, 2018-04-02:
> First off new project asks you to choose board. There is no process button (is it "finish now")

I have never created a 'new' project in Platform-IO.   I just unzip (or clone) the existing bugfix_2.0.0 branch and 'open' that.

The configuration.h files are identical between bugfix_1.1.x and bugfix_2.0.0 right now.   You should be able to drop a working set of files from one into the other.

ModMike, 2018-04-02:
Roxy,

Thank you for the help.  I opened the Marlinbugfix2.0 and built it.  It completed perfectly.

The only issue I had was when i copied my config files over and recompiled, it said:

Marlin/src/HAL/HAL_AVR/../../inc/SanityCheck.h:46:4: error: #error "You are using an old Configuration.h file, update it before building Marlin."

It is the most recent one from 1.18.  What do I need to update?

Thanks again!

ModMike, 2018-04-02:
Sorry about all the posts, but I hope it will help someone.

I defined the board as tBOARD_RAMPS_14_RE_ARM_EFB in line 136 and then selected PIO Build LPC1768.

It builds with a lot of warnings but no errors.  I feel like i'm doing something wrong.  Any clues?

ModMike, 2018-04-02:
**Re-Arm Build tutorial for total newbs like me Version 1.0**

Please note that this is a work in progress.  I will remove this warning when the process is vetted by a few people.

Works for OS X and Windows

1) Install Atom and Platform IDE by following these [instructions](url)

2) Download Marlin 2.0 by [clicking here](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip)

3) Unzip the file to your documents file (or wherever you like)

3) Open Atom.  If the home page does not open,  click on the Home icon in the upper left hand corner.

4) Click on the "Open Project" tab under the Quick Access heading in the "PlatformIO Home" screen.  From the dialog box navigate to where you unzipped the download.

5) Click on the "Marlin-bugfix-2.0.x" folder.  The bottom right dialog will become blue and read Open "Marlin-bugfix-2.0.x".  Click on it to open project

6) You can close the welcome page by hovering on the "PlatformIO Home" title and then clicking on the X that appears to the right.

7) Under Project, click on the marlin tdirectory and select configuration.h.  It will open on the right.

8) Scroll down to line 108 and change the #define SERIAL_PORT value from 1 to -1 to select USB

9) Scroll to line 117 and comment out the #define SERIAL_PORT_2 -1 by putting // in front of it so it looks likes this: **// #define SERIAL_PORT_2 -1**

10)Scroll down to line 136 and #define MOTHERBOARD to BOARD_RAMPS_14_RE_ARM_EFB (if thats the ramps config you use).

If you use another variant you can see the list of supported boards by clicking on Marlin-bugfix-2.0.x, Marlin, SRC, Core, Boards.h under the Project header on the right side.  Scroll down until you see your particular config and copy paste that to the end of the #define MOTHERBOARD on line 136.

11)  Save your config.h by clicking on file, save in the top menu bar.

**Building:**

1) Click on the check mark in the left panel or "PIO Build" in the bottom right hand corner to start the build process.

2) In the dialog box type PIO build LP and select the first option "Build PIO (LPC1768)"

3) The build window will open and the compile will start.  If it is successful, the window will close and the .bin 

**Uploading:**

If there is a properly formatted SD card in your Re-Arm board and it is powered on, you will see it in file explorer or on your Mac desktop. You do not need to remove it from your re-arm and put it in your computer, although that works too.

Make **sure** hidden files are visible on your machine because we need to access a hidden directory to get the firmware.bin file.

1) Use file explorer (or finder on Mac) and navigate to your Marlin-bugfix-2.0.x" folder. Mine is in documents so: Documents, Marlin-bugfix-2.0.x, .pioenvs, LPC1768

2) Copy the firmware.bin file to your re-arm sd card.

3) Reboot your re-arm.  You cannot just reset, you need to power cycle by removing all power and then plugging in again.

You should now be able to communicate and send commands with your terminal of choice.

ModMike, 2018-04-02:
Re-Arm Build tutorial

1) Install Atom and Platform IDE by following these [instructions](url)

2) Download Marlin 2.0 by [clicking here](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip)

3) Unzip the file to your documents file

3) Open Atom.  If the home page does not open,  click on the Home icon in the upper left hand corner.

4) Click on the "Open Project" tab under the Quick Access heading in the "Welcome To Platform" screen.  From the dialog box navigate to where you unzipped Marlin

Roxy-3D, 2018-04-02:
> My config_adv.h It is the most recent one from 1.18. The version number is different though. Line 35 the 1.18 confi_adv.h is 010107 and the new one is 020000.
> Can I just change the number or are there functional differences?

I believe so...   But it wouldn't hurt to do a visual diff of your files vs. the current files just to make sure.

> I defined the board as BOARD_RAMPS_14_RE_ARM_EFB in line 136 of stock 2.0 config.h and then selected PIO Build LPC1768.

Yes...   This will give you an extruder, a fan and a heated bed.

> It builds with a lot of warnings but no errors. I feel like i'm doing something wrong. Any clues?

There are a lot of warnings.   And those are slowly being fixed.

> Uploading:

One thing important to note here...   Once you have put the firmware.bin file on the SD memory card, and booted the Re-ARM board...   You no longer need to remove the SD memory card.   You can plug a USB cable into the board and connect it to your computer.   And just copy any subsequent firmware.bin across the USB cable.

ModMike, 2018-04-02:
Roxy,

I was trying to do that but am having issues configuring the upload_port in Platformio.  Now that I have the right target board, I will do some tests.

Roxy-3D, 2018-04-02:
I like the tutorial.   I think it would be good to add it to the #7076 thread (as that is the main 32-bit thread).

p3p, 2018-04-02:
>  But it is necessary to put the firmware.bin image on the SD memory card the first time (to get the USB drivers loaded into the Re-ARM processor).

That should not be the case, we don't change the bootloader so Marlin will just flash from the smoothieware usb disk, just as you will be able to flash straight back to smoothieware from the Marlin usb disk.

@ModMike nice tutorial, I leave have both serial ports active (the usb serial takes to long to start to get the startup messages that are useful for debugging) 

> Reboot your re-arm. You cannot just reset, you need to power cycle by removing all power and then plugging in again.

A simple reset works for me you shouldn't need to power off the board

p3p, 2018-04-02:
> I was trying to do that but am having issues configuring the upload_port in Platformio. Now that I have the right target board, I will do some tests.

PlatformIO does have the ability to copy the built firmware directly to an sd-card (usb drive), but we don't have that as the default because of the way it will try to autodetect the correct drive, We'l end up with firmware in random drives on peoples computers.

ModMike, 2018-04-02:
I cribbed the rest procedure from another tutorial.  I will update the tutorial to reflect your experience.  As to the USB, if I enable the serial ports I have massive compilation errors.

For auto upload, you need to specify the the target volume so I think it will be pretty safe.  I would much rather flash like we did with Arduionos and I've read that is possible but not sure how to do it.  will work on it.

As soon as I validate this tutorial i will post where Roxy recommneded.

ModMike, 2018-04-02:
I'm getting a lot of U*G warnings during compile:

^~
Marlin/src/HAL/HAL_LPC1768/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp:151:11: note: ...this statement, but the latter is misleadingly inden
ted as if it were guarded by the 'if'
register uint8_t *ptr = (uint8_t *)arg_ptr;

What do I need to put in the platform.ini for the target uploading?  is the following correct?

[platformio]
src_dir = Marlin
build_dir = .pioenvs
lib_dir = .piolib
libdeps_dir = .piolibdeps
env_default = LPC1768

and in the env section:

env:LPC1768]
platform        = nxplpc
board           = lpc1768
upload_port     = /Volumes/ReArm
board_f_cpu     = 100000000L
build_flags     = !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py
  ${common.build_flags}
  -DU8G_HAL_LINKS
src_build_flags = -Wall
build_unflags   = -Wall
lib_ldf_mode    = off
lib_extra_dirs  = frameworks
lib_deps        = CMSIS-LPC1768
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  TMC2130Stepper@>=2.2.1
extra_scripts   = Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py
src_filter      = ${common.default_src_filter}
monitor_baud    = 250000

It also can never fin CMSIS-LPC1768.  I get the following warning at the beginning:

Detected non-PlatformIO `//upload_port` option in `[env:LPC1768]` section
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK
). E.g., SPI, Wire, etc.

AJMOZ, 2018-04-03:
I will investigate

ModMike, 2018-04-03:
I am still not able to make this work.  I tested my setup by compiling and uploading to an atmega2560 ramps 1.4 setup.  That compiled perfectly and ran as expected.  Responds to M119 with all open when ramps 1.4 board connected and as closed when only running Atmega2560.

Now that my tool chain was validated, I did what I listed in the tutorial.  I am able to create the firmware.bin and put it on the SD card.  USB connects and echoes like nobody business.  The only problem is I get no response to an M119 command, even when endstops are jumpers or physically connected.

I used the same stock config.h that in the project, set board to  BOARD_RAMPS_14_RE_ARM_EFB and LPC1768 in platform.ini (which I forgot in my tutorial).

Questions:

1) Is this correct?
2) Am I supposed to get a response to m119 on a naked board with ramps 1.4 connected?
3) Doesn't LPC1768 need the bed environment specified?
4) I can't get the upload port to work, can someone tell me exactly where to specify the upload_port and how to upload?
5) Finally, what does the LPC1768 debug and upload config do that the LPC1768?

Thank you.

thinkyhead, 2018-04-08:
Thanks for providing the documentation, @ModMike! It's going to be extremely helpful to others, especially us who can just pass on the link!

Negaduck, 2018-06-03:
Cheers everybody. I run into some several issue which i cant solve on my own. I can get a successful Build when i follow the instructions on marlinfw.org for the Re-arm Board. The first problem is when i upload the file onto the fat32 formatted sd card(32gb) and after press the reset button i simply cant Connect pronterface. The Message Connecting... comes but it stays. 
Then i tried to compile it with lcd screen enabled but the linked u8glib-hal throws out errors about Class definitions within this library. 
What can i do to get one of them working? 
I Hope somebody is able to help me. I just dont want to install smoothieware haha.

p3p, 2018-06-03:
Can you create an issue following the template (supply your configs, exact error logs) thanks. Just saying you have errors isn't enough info to help.

TwatBurglar, 2018-10-02:
I'm having trouble reflashing:    I stuff firmware.bin onto Rearm's SDcard (rearm is mounted in linux via USB)....   but even a powercycle (including USB unplug, so there's no power leaking through) doesn't reliably get me the magical FIRMWARE.CUR on the next reboot.

The only reliable way of turning firmware.bin -> FIRMWARE.CUR  is pressing the reset button on RAMPS.  Unfortunately, the hardware is entombed deep in the bowels of a steel enclosure... making button-pressing impossible.

Anyone have any ideas?

ModMike, 2018-10-02:
Is your card larger than 32GB?  If so partition it into 2 partitions.  Here is the guide I wrote:

http://marlinfw.org/docs/basics/install_rearm.html

ModMike, 2018-10-02:
You can do a full power down to reset re-arm.  Simply running your printers power supply off is good enough.

p3p, 2018-10-02:
@ModMike it has never needed 2 restarts the bootloader just copies the binary it finds on boot and runs it, on either reset or power cycle as long as it turns off. The LEDS are also controlled by Marlin now so you will get a constant flashing status LED while Marlin is running.

@TwatBurglar There is nothing special about the reset button, the important thing is to power cycle.

ModMike, 2018-10-02:
There has been a lot of discussion around this and it's not always clear.  I also may be conflating the restart with the Azteeg X5 Mini Wifi which absolutely requires 2 restarts.  First one compiles and LEDs sweep, ending up with 2 green and 1 red on.  Second restart, all green.

p3p, 2018-10-02:
That is because of default state the bootloader leaves the LEDs in after a flash (it shows the sweep progress bar), as long as you are using the smoothieware bootloader then 1 reboot is always enough, If your board has a define for an LED pin in its pins file then Marlin will start to flash it as a status indicator otherwise they will be left in the state the bootloader leaves them, I made some assumptions based on smoothie and turn of the "other" unused LEDs.

TwatBurglar, 2018-10-02:
@p3p @ModMike  PowerCycling doesn't work for me...  when the power turns back on and ReArm boots,  firmware.bin is sitting pretty, untouched.  And laughing at me.

Upon buttonpress, firmware.bin turns into FIRMWARE.CUR,  first time, every time.

Something is fishy (on my system, at least):  16gb fat32 SD card;  tried mounting over usb on both linux and windows (no, windows/octoprint is not caching firmware.bin or some such) ;  checked for power leakage during poweroff...  ???

ModMike, 2018-10-02:
Gotcha.  By the way, any progress on Marlin using the onboard SD?

p3p, 2018-10-02:
@TwatBurglar if the board turns off (gets reset) then it searches the card (as long as it can access it) for a bin file and flashes it on the next boot.

@ModMike Gloomyandy has that implemented in the new [framework](https://github.com/p3p/Marlin/tree/pr_bf2_pulloutlpcframework) (if you test it give me feedback through discord), I'm hoping to get a PR so marlin starts using it soon(tm), keep getting hung up on (important) details

p3p, 2018-10-02:
@ModMike Also you should set an LED pin in the pins files if the board has them so the pins are initialised.

ModMike, 2018-10-02:
Ok will do.