MarlinFirmware/Marlin #7797: Issue when using BABYSTEP_ZPROBE_GFX_OVERLAY

dot-bob, 2017-09-30:
@marcio-ao,
If I enable the option #define BABYSTEP_ZPROBE_GFX_OVERLAY in configuration.h with the latest bugfix-1.1x branch I am getting the following errors (see my Travis CL output): [https://travis-ci.org/dot-bob/Marlin/builds/280716639?utm_source=github_status&utm_medium=notification](https://travis-ci.org/dot-bob/Marlin/builds/280716639?utm_source=github_status&utm_medium=notification)

Am I missing something that is causing these errors?

This seems to happen on all my builds derived from the latest bugfix-1.1.x as of today.  Perhaps it should be added it to the travis build check.

Roxy-3D, 2017-09-30:
I think you found a bug!   We should @ that PR's author.

Tannoo, 2017-10-01:
I ran into the same thing in `bugfix-2.0.x`. I couldn't seem to get the bitmaps declared. So....

Cut this from `dogm_bitmaps.h`:

```
#if ENABLED(BABYSTEP_ZPROBE_GFX_OVERLAY)
  const unsigned char cw_bmp[] PROGMEM = { //AVR-GCC, WinAVR
    0x07,0xf8,0x00, // 000001111111100000000000
    0x0c,0x0c,0x00, // 000011000000110000000000
    0x10,0x02,0x00, // 000100000000001000000000
    0x20,0x01,0x00, // 001000000000000100000000
    0x60,0x01,0x80, // 011000000000000100000000
    0x40,0x00,0x80, // 010000000000000010000000
    0x40,0x03,0xe0, // 010000000000001111100000
    0x40,0x01,0xc0, // 010000000000000111000000
    0x40,0x00,0x80, // 010000000000000010000000
    0x40,0x00,0x00, // 010000000000000000000000
    0x40,0x00,0x00, // 010000000000000000000000
    0x60,0x00,0x00, // 011000000000000000000000
    0x20,0x00,0x00, // 001000000000000000000000
    0x10,0x00,0x00, // 000100000000000000000000
    0x0c,0x0c,0x00, // 000011000000110000000000
    0x07,0xf8,0x00  // 000001111111100000000000
  };

  const unsigned char ccw_bmp[] PROGMEM = { //AVR-GCC, WinAVR
    0x01,0xfe,0x00, // 000000011111111000000000
    0x03,0x03,0x00, // 000000110000001100000000
    0x04,0x00,0x80, // 000001000000000010000000
    0x08,0x00,0x40, // 000010000000000001000000
    0x18,0x00,0x60, // 000110000000000001100000
    0x10,0x00,0x20, // 000100000000000000100000
    0x7c,0x00,0x20, // 011111000000000000100000
    0x38,0x00,0x20, // 001110000000000000100000
    0x10,0x00,0x20, // 000100000000000000100000
    0x00,0x00,0x20, // 000000000000000000100000
    0x00,0x00,0x20, // 000000000000000000100000
    0x00,0x00,0x60, // 000000000000000001100000
    0x00,0x00,0x40, // 000000000000000001000000
    0x00,0x00,0x80, // 000000000000000010000000
    0x03,0x03,0x00, // 000000110000001100000000
    0x01,0xfe,0x00  // 000000011111111000000000
  };

  const unsigned char up_arrow_bmp[] PROGMEM = { //AVR-GCC, WinAVR
    0x06,0x00, // 000001100000
    0x0F,0x00, // 000011110000
    0x1F,0x80, // 000111111000
    0x3F,0xC0, // 001111111100
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00  // 000001100000
  };

  const unsigned char down_arrow_bmp[] PROGMEM = { //AVR-GCC, WinAVR
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x06,0x00, // 000001100000
    0x3F,0xC0, // 001111111100
    0x1F,0x80, // 000111111000
    0x0F,0x00, // 000011110000
    0x06,0x00  // 000001100000
  };

  const unsigned char offset_bedline_bmp[] PROGMEM = { //AVR-GCC, WinAVR
    0xFF,0xFF,0xFF // 111111111111111111111111
  };

  const unsigned char nozzle_bmp[] PROGMEM = { //AVR-GCC, WinAVR
    0x7F,0x80, // 0111111110000000
    0xFF,0xC0, // 1111111111000000
    0xFF,0xC0, // 1111111111000000
    0xFF,0xC0, // 1111111111000000
    0x7F,0x80, // 0111111110000000
    0x7F,0x80, // 0111111110000000
    0xFF,0xC0, // 1111111111000000
    0xFF,0xC0, // 1111111111000000
    0xFF,0xC0, // 1111111111000000
    0x3F,0x00, // 0011111100000000
    0x1E,0x00, // 0001111000000000
    0x0C,0x00  // 0000110000000000
  };
#endif // BABYSTEP_ZPROBE_GFX_OVERLAY
```

And insert in in the bottom of `ultralcd.h` and things will be good.

apballard, 2017-10-04:
Yeah also having this issue.

Tannoo, 2017-10-18:
BTW, @cdedwards gave me the correct correction. I thought I would update that here.

closed 2017-10-25 by dot-bob