MarlinFirmware/Marlin #7800: [FR] Change height for manual delta calibration menu

Astroman, 2017-09-30:
Manual delta calibration in the LCD menu sends the height to 10mm every time one of the positions is selected. This is annoying high since the position has to be sent down in small steps every time. This used to be 4mm and it was clear where to change it in previous versions. Is it possible to change this to something smaller? Thanks.

thinkyhead, 2017-11-15:
@LVD-AC

LVD-AC, 2017-11-15:
The moves in manual delta calibration are not straight but go down in the middle ???
Anything lower will scrape the bed.

LVD-AC, 2017-11-15:
I've checked this on my machine: it lowers with delta_height - 2mm from the home position. Increase your delta_height to be closer to bed.

closed 2018-02-09 by thinkyhead