MarlinFirmware/Marlin #7801: Screen is stuck when m600 with dual extruder

reisher, 2017-09-30:
### Bug Report
- Description: ---
When I run m600 command from my screen aka change filament the print stops, beeps, axis are moving and it unloads the filament. I hit ok to resume and then I get stuck on heating nozzle 240/240 forever until I unplug the printer no matter what I do.
I suspect it's caused by having the second extruder being missing from the m600 feature somehow? The target temperature is saved from the inactive extruder and resumed on the active one? I tried to find something in the Marlin_main.cpp and I saw where it's storing it's just using "e" for extruder.
[Uploading Marlin-1.1.x.zip…]()

- Expected behaviour: ---
To have it change filament and let me resume my print

- Actual behaviour: ---
it's stuck heating even tho its already at the target temperature

thinkyhead, 2017-11-15:
Still seeing this is the latest `bugfix-1.1.x`?

boelle, 2019-02-19:
@reisher Still seeing this is the latest `bugfix-2.0`?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead