MarlinFirmware/Marlin #7804: Temperature reports approx 70 degrees higher than normal

kruuth, 2017-10-01:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: --- temperature is returning 90-100 degrees on bed.  When you attempt to heat the bed beyond this temperature it behaves normally.  However, the display is always registering values between 93 and 100 degrees.  Setting the value to anything below this results in the printer thinking that the bed temperature is already high.  With no heating at all, using ambient temperature, it should be reporting 25-27 degrees.

- Expected behaviour: --- bed should be reporting around 25 degrees ambient
- Actual behaviour: --- bed reports 90-100 degrees ambient temperature
- Steps to reproduce:  Download and install bugfix1.1.x.  Copy Anet A8 Configuration.h and Configuration_adv.h.   
  - Enable inductive sensor
  - Compile as normal

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1346933/Marlin.zip)

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

kcheeeung, 2017-10-01:
You may want to check if you have the right thermistor profile in your config file. That is my first guess as to why the temperature looks like it's wrong.

kruuth, 2017-10-01:
That's what I thought at first, however I'm using the correct settings for the default A8 bed.

Tannoo, 2017-10-01:
Bad connection?

kruuth, 2017-10-01:
No, It reports fine at higher temperatures.

Tannoo, 2017-10-01:
Maybe it is a wrong default thermocouple.

kruuth, 2017-10-01:
Do you know what the default is for the Anet A8?  It is presently set for 100K thermistor - ATC Semitec 104GT-2 (Used in ParCan & J-Head) (4.7k pullup), which is in the default config.

Tannoo, 2017-10-01:
I don't. Just try another one. 100k... try #1.

kruuth, 2017-10-01:
I did.  It's returning 112 now.

kcheeeung, 2017-10-01:
A8 is your **printer.** The manufacturer may provide different thermistors, and you have to find out which one.

kruuth, 2017-10-01:
I've gone through the entire list in the config file.  Nothing.  The lowest I got it was 87.

kruuth, 2017-10-01:
If it helps, this is the Anet A8 from Gearbest.

kruuth, 2017-10-01:
Ok.  I've gone through the entire list.  I've gotten as high as 550 and as low as 50 but it's still much higher than ambient temperature.

Tannoo, 2017-10-01:
I would tend to believe a bad thermo, or bad connection.

I haven't that issue with mine accross many different versions of Marlin.

Tannoo, 2017-10-01:
Mine is not an A8 though... maybe a board issue?

AnHardt, 2017-10-02:
Connect the thermistor to the pin for the second extruder. Swap the pins in the pin-file.
If you get the same readings, it's reasoned by thermistor or cabling (or software error). If not, by the board.

kruuth, 2017-10-02:
Ok.  Tried that.  Same issue.  Replaced the board and it's working.  Chalk this up to a bad board.

closed 2017-10-02 by kruuth