MarlinFirmware/Marlin #7806: Marlin-bugfix-2.0.x compile error

vMeph, 2017-10-01:
after update Marlin-bugfix-2.0.x from last night  i cant compile now it gives me this error
any ideas

In file included from C:\Users\vMeph\AppData\Local\Temp\arduino_build_738152\sketch\src/inc/MarlinConfig.h:33:0,

                 from C:\Users\vMeph\Desktop\Marlin-bugfix-2.0.x\Marlin\Marlin.ino:31:

C:\Users\vMeph\AppData\Local\Temp\arduino_build_738152\sketch\src/inc/../HAL/HAL.h:32:25: fatal error: src/inc/SPI.h: No such file or directory

 #include "src/inc/SPI.h"

                         ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

thanks

bobc, 2017-10-01:
Change that line to 
```#include "../../src/inc/SPI.h"```

vMeph, 2017-10-01:
that worked but now i got more diferent errors

n file included from C:\Users\vMeph\AppData\Local\Temp\arduino_build_27661\sketch\src\lcd\ultralcd_impl_DOGM.h:46:0,

                 from C:\Users\vMeph\AppData\Local\Temp\arduino_build_27661\sketch\src\lcd\ultralcd.cpp:87:

C:\Users\vMeph\AppData\Local\Temp\arduino_build_27661\sketch\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:26:24: fatal error: src/Marlin.h: No such file or directory

 #include <src/Marlin.h>

                        ^

compilation terminated.

Using library U8glib at version 1.19.1 in folder: C:\Users\vMeph\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

is this a bug or just happens to me?
cause before last night all worked ok

bobc, 2017-10-01:
Change that line to

```#include "../../../src/Marlin.h"```

These problems were introduced by https://github.com/MarlinFirmware/Marlin/pull/7746 The problem is that include paths that work for PlatformIO doesn't necessarily work in Arduino IDE, because they have different rules for include paths.

vMeph, 2017-10-01:
@bobc  it keep always trowing error so i downloaded platform ide and all compile great

i got is a issue with the controller fan i have a open issue but never no one replyed if would it be possible take a quick look the link is here
https://github.com/MarlinFirmware/Marlin/issues/7791

thanks alot for all the help

closed 2017-10-04 by vMeph